purduesigbots/pros #4: pros-install_ubuntu.sh has a poor user experience

catb0t, 2016-12-09:
The [pros-install_ubuntu.sh](https://github.com/purduesigbots/pros/releases/download/2.11.0/pros-install_ubuntu.sh) script installed Atom v1.10.2 over my previous v1.12.6. Downgrading the LSD in a minor version breaks some small functionality, but more importantly I'd like to keep my latest version! 

Moreover, the script installed the Atom extension `tool-bar` which appears to be (mostly) unrelated to PROS proper but  doesn't *ask* to do these things. `echo "Install Atom extensions?"; read -n2 answer` would have been better.

Finally, I think it is poor code quality when a non-system program asks me for my Linux password. For all I know, that `[sudo] password for cat:` prompt is completely fake and going to send my password to your servers so you can add me to your botnet.

Instead of running `sudo apt-get update ; sudo apt-get install ...`, use something like this:

```
if [ "$EUID" -ne 0 ]
  then echo "Please run as: sudo $0"
  exit
fi
apt-get update
apt-get install ...
```

I'd open a pull request except I can't because the `pros-install_ubuntu.sh` script is release-only.

HotelCalifornia, 2016-12-09:
The `tool-bar` extension is used to provide a graphical interface to some of the pros CLI commands.

catb0t, 2016-12-10:
@HotelCalifornia  I understand, but I'd like to be asked before installing things that fundamentally change my development environment, that I was not expecting.

edjubuh, 2017-01-01:
Hi @catb0t,

You're right, we shouldn't accidentally downgrade your installation of Atom (ideally it'd always try to install the latest if you don't presently have Atom installed). If you have an idea for how to implement this functionality, feel free to post it here and we'll integrate with our next release.

We strive to create a consistent initial development environment for all users upon installation. Many of our users are high school students who are learning how to program and may not be as familiar with using facilities like the Command Palette. I'm positive having the tool-bar to provide facilities for quickly creating new files, uploading binaries, and opening a console are much appreciated by these users. Experienced users of Atom (who are familiar with Command Palette and don't have a use for the tool-bar) will understand how to uninstall tool-bar without issues. It makes more sense to install by default for our purposes.

Your final point about not running `sudo` for our users is partially valid. While it can be considered a hazard to input your sudo password, the script you run is openly verifiable - just print it out and see that the script isn't malicious. That being said, it is standard practice to make the whole script be run with `sudo`, so we'll update the script accordingly.

catb0t, 2017-03-04:
That's excellent, thanks for taking the time to look into my suggestions!

HotelCalifornia, 2017-03-31:
TODO

mLopez3233, 2017-04-19:
Well here is my first post and first shell script.

For example, are you people thinking something like this?

* will not touch your atom installation
* if you dont have atom, it will install the lastest version
*should work for other ubuntu versions, but I have 17.04 so its only tested there

[pros-install_ubuntu-17.04.txt](https://github.com/purduesigbots/pros/files/933492/pros-install_ubuntu-17.04.txt)

HotelCalifornia, 2017-04-20:
For continuity, here's the script proposed by @mLopez3233:
```
#!/bin/bash

USER=`whoami`
ATOM_URL='https://atom.io/download/deb'
DOCUMENTATION_URL='pros.cs.purdue.edu/api/'
ISSUE=`cat /etc/issue`

if [ "$USER" = "root" ]; then
  continue
else
  echo "Please run as root"
  exit
fi

read -p "Type 'yes' to continue? " choice
if [ "$choice" = "yes" ]; then
  echo "Okay, checking some more things"
  echo
else
  echo "Operation cancelled by user"
  exit
fi

if [ "$ISSUE" = "Ubuntu 17.04 \n \l" ]; then
    echo "Ubuntu 17.04 detected"
else
  echo "This is not Ubuntu 17.04"
  read -p "Try to install anyways? [N|y] " choice
  if [ "$choice" = "y" ]; then
    echo
  else
    echo "Operation cancelled by user"
    exit
  fi
fi

echo "Ready for installation"
echo

echo "PROS UBUNTU INSTALL SCRIPT"
echo "This script will install Pros and all of its core dependencies."
echo

echo "Installing environment dependencies: git, arm-none-eabi-gcc, clang, python3, python3-pip"
echo

sudo apt -q update
sudo apt -y install git clang gcc-arm-none-eabi python3 python3-pip

if [ $? -ne 0 ]; then
    echo "Something went wrong in linux dependency installation"
    exit 1
fi

command -v atom >/dev/null 2>&1 || ATOM="true"; }

if [ "$ATOM" = "true" ]; then
  echo
  echo "Atom installation has been detected"
  echo "therefore Atom will not install"
else
  echo "Installing Atom"

  wget $ATOM_URL
  if [ $? -eq 0 ]
  then
      sudo dpkg -i atom-amd64.deb
      sudo apt-get install -f
      rm -f atom-amd64.deb
  else
      echo "Failed to download Atom installation from github"
      exit 1
  fi
fi

echo "Installing python PROS CLI"
sudo -H pip3 install pros-cli

if [ $? -ne 0 ]
then
    echo "pros-cli failed to install open an issue on github for support"
    exit 1
fi

echo "Installing PROS atom plugins"
echo
apm install platformio-ide-terminal linter-ui-default intentions busy-signal file-icons linter tool-bar tool-bar-main busy build pros

if [ $? -ne 0 ]
then
    echo "Atom package installations have failed"
    exit 1
fi

echo "========================================================"
echo "Install will mostly  likely work, ready to start programming your cortex."
echo "Check out $DOCUMENTATION_URL for all PROS documentation"
echo "========================================================"

```

HotelCalifornia, 2017-04-20:
Thanks for your input. My only major issue with the script you've offered is the check for Ubuntu 17.04, as I feel the vast majority of Ubuntu users are running on older versions, and it's not fair to make these users feel like they're doing something that might break. That being said, we will run tests on some older Ubuntu installs to verify everything works properly (but I see no reason to think it won't).

Other than that, it looks good, and we'll include this in our next installer.
:+1:
.

HotelCalifornia, 2017-04-20:
I've made some minor modifications, including making the language a little more consistent, and removing redundant `sudo` (and the 17.04 check). 

```bash
ER=`whoami`
ATOM_URL='https://atom.io/download/deb'
DOCUMENTATION_URL='pros.cs.purdue.edu/api/'

if [ "$USER" = "root"  ]; then
        continue
else
        echo "Please run as root"
        exit
fi

echo "PROS UBUNTU INSTALL SCRIPT"
echo "This script will install PROS and all of its core dependencies."
echo
read -p "Type 'yes' to continue: " choice
if [ "$choice" -ne "yes"  ]; then
        echo "Operation cancelled by user"
        echo
        exit
fi

echo "Installing environment dependencies: git, arm-none-eabi-gcc, clang, python3, python3-pip"
echo

apt -q update
apt -y install git clang gcc-arm-none-eabi python3 python3-pip

if [ $? -ne 0  ]; then
        >&2 echo "Failed to install apt dependencies"
        echo
        exit 1
fi

command -v atom >/dev/null 2>&1 || ATOM="true";

if [ "$ATOM" = "true"  ]; then
        echo
        echo "Atom installation has been detected"
        echo "Skipping Atom installation..."
else
        echo "Installing Atom"

        wget $ATOM_URL
        if [ $? -eq 0  ]
        then
                sudo dpkg -i atom-amd64.deb
                sudo apt-get install -f
                rm -f atom-amd64.deb
        else
                >&2 echo "Warning: Failed to download Atom deb package from $ATOM_URL!"
                exit 1
        fi
fi

echo "Installing python PROS CLI"
sudo -H pip3 install pros-cli

if [ $? -ne 0  ]
then
        >&2 echo "Warning: Failed to install pros-cli!"
        exit 1
fi

echo "Installing PROS atom plugins"
echo
apm install platformio-ide-terminal linter-ui-default intentions busy-signal file-icons linter tool-bar tool-bar-main busy build pros

if [ $? -ne 0  ]
then
        >&2 echo "Warning: Failed to install requisite Atom packages!"
        exit 1
fi

echo "============================================================================"
echo "*  Installation finished, you are ready to start programming your cortex.  *"
echo "*  Check out $DOCUMENTATION_URL for all PROS documentation.                *"
echo "============================================================================"
```

mLopez3233, 2017-04-20:
> Thanks for your input. My only major issue with the script you've offered is the check for Ubuntu 17.04, as I feel the vast majority of Ubuntu users are running on older versions, and it's not fair to make these users feel like they're doing something that might break. That being said, we will run tests on some older Ubuntu installs to verify everything works properly (but I see no reason to think it won't).  @-HotelCalifornia

To be clear I only included the 17.04 test because I only have 17.04 and I don't know if it will work in older version. However, I see no reason why it would not work.

Also, did you notice the extra atom plug ins, i think they are :linter-ui-default intentions busy-signal: ---in short, saves the user from an extra step of clicking install plug in decencies when atom fires up. Just imagine how tiering that could be.

PS, I really enjoy your 'minor editing'.

HotelCalifornia, 2017-06-20:
@catb0t I just realized that the proposed install script only announces when components will be installed, rather than prompting the user, which is what you originally asked for. However, the pros-atom package will break if it does not have its dependencies installed (namely, packages like tool-bar). What are your thoughts on this?

One option would be to remove that step from the script, and have Atom prompt users to install dependencies on first startup.

HotelCalifornia, 2017-10-14:
https://github.com/purduesigbots/pros-cli/tree/2.6.0

closed 2017-10-14 by HotelCalifornia