purduesigbots/pros #5: C++ newlib compatibility

PixelToast, 2016-12-31:
PROS and newlib have a conflicting FILE symbol
`In file included from ../include/main.h:46:0,
                 from lcd.cpp:4:
../include/API.h:744:13: error: conflicting declaration 'typedef int FILE'
 typedef int FILE;
             ^
In file included from /usr/include/newlib/c++/5.4.1/cwchar:44:0,
                 from /usr/include/newlib/c++/5.4.1/bits/postypes.h:40,
                 from /usr/include/newlib/c++/5.4.1/bits/char_traits.h:40,
                 from /usr/include/newlib/c++/5.4.1/string:40,
                 from /usr/include/newlib/c++/5.4.1/stdexcept:39,
                 from /usr/include/newlib/c++/5.4.1/array:38,
                 from /usr/include/newlib/c++/5.4.1/tuple:39,
                 from lcd.cpp:1:
/usr/include/newlib/wchar.h:53:16: note: previous declaration as 'typedef __FILE FILE'
 typedef __FILE FILE;
                ^
`
this can be solved by either changing the name of `FILE` or changing all the `#define`s to use `static const` instead and putting all the pros API functions into a namespace

edjubuh, 2017-01-09:
Maintaining symbol compatibility with every library is quite an undertaking. The official PROS 2.0 kernel will respect libc and GCC constraints. I'll leave it open if someone from the community decides to modify the filesystem, but the PROS development team don't currently have plans to address this for PROS 2.0.

iuyte, 2017-06-28:
I fixed this with CTRL+F and replacing all instances of FILE with PROS_FILE in API.h, as per https://blog.pxtst.com/using-c-in-pros/.

commit cfbb14f7 (2017-06-30): Add C++ compatability with FILE symbol

FILE was renamed to PROS_FILE. If the FILE symbol wasn't defined,
then backwards compatability is maintained with #def'ing FILE to
PROS_FILE.

Fixes #5
commit cdd33312 (2017-07-09): Improve C++ compatability with FILE symbol

Rename FILE to PROS_FILE. If FILE isn't defined, the FILE gets aliased
to PROS_FILE to maintain backward compatability.

Squash merge of the following commits:

commit a85ef838c7ad0e020b4b5b8df79ff41c40b52226
Author: Alex Brooke <akb.sbc@gmail.com>
Date:   Thu Jul 6 22:19:06 2017 -0500

    update wording on the alias

    finally found wording I like

commit 82cb99743abdaa148127c48d36199f659927b400
Author: Elliot Berman <berman5@purdue.edu>
Date:   Thu Jun 29 22:10:09 2017 -0500

    Add docstring for FILE vs. PROS_FILE usage

commit cfbb14f7872903d [... cut]
closed 2017-07-22 by edjubuh