purduesigbots/pros #15: Windows installation doesn't work well for non-admin user.

dnj12345, 2017-04-30:
#### Expected Behavior:
Expected it to work for admin and non-admin user.

#### Actual Behavior:
For non-admin user, it always complains about PROS Cli not being in path and things don't work i.e. can't compile etc.

#### Steps to reproduce:
1. Install from a non-admin or admin account. Installing from non-admin account gives apm errors (that's another issue)
2. Try to access it from non-admin account. U'll see the Pros CLI not in path on the Pros Welcome page and things don't work after that. U can view and edit files but that's about it.

#### System information:
Kernel Version:
Windows 10.

#### Additional Information
I've done multiple installs. From admin accout and non-admin account. The issue is the same no matter what. I can use it fine from admin account. I want my kid to use it on his account which doesn't have admin privileges.

barinb1, 2017-04-30:
If I remember correctly, the installer puts the PROS CLI on the user's %PATH%, not the system %PATH%. As a workaround, you can add the PROS CLI to the system %PATH%.

dnj12345, 2017-04-30:
Awesome! your suggestion worked.

edjubuh, 2017-04-30:
Hi @dnj12345, I'm going to reopen this issue as we're to address our installers across the board to provide a simpler installation experience that can support admin, non-admin, all-user, and per-user installations.

dnj12345, 2017-05-01:
ok. Thanks.

For what it's worth, I had to add these two paths for a non-admin user to get things working.

c:\Program Files (x86)\Pros\cli
c:\Program Files (x86)\Pros\toolchain\bin

barinb1, 2017-05-01:
> ok. Thanks.
> 
> For what it's worth, I had to add these two paths for a non-admin user to get things working.
> 
> c:\Program Files (x86)\Pros\cli
> c:\Program Files (x86)\Pros\toolchain\bin
I'm pretty sure you _technically_ only need one of the two entries (Atom will complain but you can still compile and download code)

dnj12345, 2017-05-01:
With the first path, I was able to get past the error on the Pros Welcome page i.e. the error that says (in red).. "Pros CLI not found. Make sure Pros CLI is in you path"

Without the second path, I was not able to compile. Without it, I was getting the error... 'make' not found

barinb1, 2017-05-01:
> With the first path, I was able to get past the error on the Pros Welcome page i.e. the error that says (in red).. "Pros CLI not found. Make sure Pros CLI is in you path"
> 
> Without the second path, I was not able to compile. Without it, I was getting the error... 'make' not found

`...\cli` will henceforth be referred to as the CLI
`...\toolchain\bin` will henceforth be referred to as the toolchain

To clarify what I meant: Atom is just about the only thing that cares about the CLI. Without it, you will still be able to compile and download with `make -k upload`. I forgot the dependency of the CLI on the toolchain being in `%PATH%`, so my previous statement was wrong; the amended statement is _you only need the second entry technically_.

dnj12345, 2017-05-01:
That's probably true. @edjubuh reopened the issue to address all the issues related to the installers. My comment was intended to provide the team as much info as possible.

sanyaade-artificial-intelligence, 2017-06-25:
The main issue is to address is to make the installer to put everything that is PROS in one folder e.g. c:\pros. Also avoid path with spaces

Also allow user to be able to specify destination directory/custom folder. 

The above will make PROS user friendly, portable to run run from USB flash memory, etc... just like the Eclipse based edition..

Please also add to the documentation section on how build the IDE from source code on 3 major platforms like Windows, Linux and OSx. 

Many thanks!

God blesses!!!

Best regards,
Sanyaade

HotelCalifornia, 2017-06-25:
Thanks for your input.

[PROS for Atom](https://github.com/purduesigbots/pros-atom) is a package for the [Atom text editor](https://github.com/atom/atom), so if you really want documentation on building the editor from source, you should consult their documentation.

sanyaade-artificial-intelligence, 2017-06-25:
@HotelCalifornia

I am talking about PROS compiler in general, the way it installs pm Windows platform that has led to the above stated problems by other users above. Also the installation/compiling I ask for is about PROS and not the atom Editor as I believe that any editor can be configured to use with PROS.

I am trying it with YARGATO and then CodeBlock IDE when I get the PROS command-line edition working. Also need to see if I could use it with Vex IQ library

many thanks for your quick reply! Cheers!

God blesses!!!

Best regards,
Sanyaade

edjubuh, 2017-06-25:
> The main issue is to address is to make the installer to put everything that is PROS in one folder e.g. c:\pros.

This is presently possible. Click "custom" installation and you'll be prompted to change the install location. 

> Also avoid path with spaces

I can't fathom why this would be a "feature" and not a limitation

> Also the installation/compiling I ask for is about PROS and not the atom Editor as I believe that any editor can be configured to use with PROS.

I'm not sure what you're asking for. This capability is already enabled in the Windows PROS installer - just customize your installation and you can remove Atom entirely.

Please note that VEX IQ is not supported by PROS.

sanyaade-artificial-intelligence, 2017-06-25:
Sometimes its difficult to explain but see:

Atom folder on Windows:
C:\Users\Homework\AppData\Local\atom\Update.exe --processStart "atom.exe"

PROS:
C:\Users\Homework\AppData\Roaming\PROS
C:\Program files\Purdue\PROS

All these can in a single installed folder. Make it user friendly, portable and appeal to those ordinary user. As for people like you and me that can work down to systems level, this is no problem as we could just clone, fork and hack to get things done/twisted

Re: VexIQ and PROS: Yes I know it is not supported but it is portable okay so I am using JPearman documented code as a starting point.  I know that RobotC and Texas Instrument Code Composer and Energia and PROS all contain I2C definition files (.h) already. I am looking at FreeRTOS documents along.

God blesses!!!

Best regards,
Sanyaade

edjubuh, 2017-06-26:
Part of my goal with the new installers (for all platforms) will to roll our own Atom distribution with the PROS plugins pre-installed. This could remove Atom from Local App Data. We do not have any plans to allow moving away AppData\Roaming\PROS, but this would be a straightforward modification to support some environment variable or logic that you can feel free to submit a PR for.

HotelCalifornia, 2017-10-14:
https://github.com/purduesigbots/pros-cli/tree/2.6.0

closed 2017-10-14 by HotelCalifornia