purduesigbots/pros #27: Atom doesn't see CLI on Crouton

DarkMatterMatt, 2017-07-05:
#### Expected Behavior:

Atom should detect PROS CLI 

#### Actual Behavior:

Atom says "PROS CLI threw an error before returning the version."
Going into console then typing `pros` gives the following error: `RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment.` See http://click.pocoo.org/python3/#python-3-surrogate-handling for details.

Adding this to `~/.bashrc` or `/etc/profile` fixes the problem from terminal (running `pros --version` works)

    export LC_ALL=C.UTF-8
    export LANG=C.UTF-8

#### System information:
Kernel Version: `3.18.0-14861-g70c266fc4519`
OS: Crouton
Model: HP 11 G5 (4GB RAM version)

#### Additional Information

I think that this may be an Atom issue because it seems like Atom isn't seeing the environmental variables set in `~/.bashrc`.

HotelCalifornia, 2017-07-05:
Does the issue still occur if you open Atom from the terminal?

HotelCalifornia, 2017-07-05:
related: https://github.com/platformio/platformio-atom-ide/issues/112

DarkMatterMatt, 2017-07-06:
The issue still occurs if I open Atom from terminal (both `atom` and `atom -safe` have the same issue).

**Following the instructions in the issue linked:**
`rm -rf ~/.atom` and then reinstalling the packages doesn't change anything.
There is no `~/.atom/packages/platformio-ide` (but there is `platformio-ide-terminal`). There is no `lib/main.js` in `platformio-ide-terminal`.

edjubuh, 2017-07-06:
Sorry for the unclear instructions. Try modifying `~/.atom/packages/pros/main.coffee` at L34 to add `process.env.LC_ALL = 'en_US.UTF-8';`

HotelCalifornia, 2017-07-06:
To clarify on the discrepancy: the linked issue was from platformio/platformio-atom-ide whereas our package depends on platformio/platformio-atom-ide-terminal.
The solution itself _should_ still apply if we export from within our package though.

HotelCalifornia, 2017-07-06:
Looks like this is an issue about the Atom package

DarkMatterMatt, 2017-07-06:
`pros/main.coffee` does not exist, but there is a file at `pros/lib/main.coffee`. I have tried adding `process.env.LC_ALL = 'en_US.UTF-8';` after lines 33, 34 & 35 and restarting Atom each time results in the same error.
My line 34 says `@subscriptions = new CompositeDisposable`. Am I adding after the correct line?

In `lib/proscli.coffee`, lines 26, 27, 28 say

    env = process.env
    env['LC_ALL'] = 'en_US.utf-8'
    env['LANG'] = 'en_US.utf-8' 

I tried changing the capitals for utf/UTF (just in case) but nothing changed.

Also, I have confirmed that it is the same click ASCII error by adding `console.log {c, o}` to line 174 of `lib/views/welcome.coffee` and viewing the logged object. Additional information at the bottom of the log was `Click discovered that you exported a UTF-8 locale but the locale system could not pick up from it because it does not exist.  The exported locale is "en_US.utf-8" but it is not supported.` This is weird because `~/.bashrc` and `/etc/profile` both have `export LC_ALL=C.UTF-8` (not `en_US.utf-8`).

I noticed that in `lib/proscli.coffee` there are lines that set it to `en-US.utf-8`. Changing those lines to

    env['LC_ALL'] = 'C.UTF-8'
    env['LANG'] = 'C.UTF-8' 

makes proscli work 😄 Note that the capitals _do_ matter, I first tried with lowercase utf and it had the same error.

Thanks for the help 👍

DarkMatterMatt, 2017-07-06:
I have now noticed a handy "copy output" button which means I didn't need to go and console.log the output... 😔  Maybe you could rename the button to "copy traceback" or "copy error message" as opposed to output? Just a suggestion 😸

HotelCalifornia, 2017-07-06:
The error popup is a feature of the Atom text editor itself, and we don't currently plan on changing it. You may wish to open an issue over at https://github.com/atom/atom/issues/ about it, however.

edjubuh, 2017-07-06:
Bug's probably here: https://github.com/purduesigbots/pros-atom/blob/master/lib/proscli.coffee#L105

Will double check that tonight.

edjubuh, 2017-07-08:
@DarkMatterMatt the PR on pros-atom should work for you by adding a configuration option on the plugin page.

HotelCalifornia, 2017-07-08:
[Atom doesn't inherit the `LANG` environment variable](https://github.com/atom/atom/issues/4126#issuecomment-200427499), so the best we can do (without adding unmaintainable hacks) is to add a configuration option, so everyone can set their preferred locale themselves.

closed 2017-07-09 by edjubuh