purduesigbots/pros #31: Unit tests

PixelToast, 2017-07-24:
Unit tests are great for finding bugs and regressions, but these may be hard to automate

I propose linking all the unit tests together into a single binary, have a script automatically download this to a cortex, select the unit test using debug stream, reboot the cortex using the bootloader if it gets stuck, and repeat this for every unit test until they all pass or one fails

HotelCalifornia, 2019-09-19:
I think it's time to revisit this again. However, instead of actually uploading to a uC, it would be more feasible to mock the vexOS SDK and run the tests as part of CI, similar to the way OkapiLib mocks the PROS API.

Octogonapus, 2019-09-19:
I think that system tests (i.e. with the real hardware in the loop) are a vital component of a comprehensive test suite. PIO has an interesting approach to unit testing on hardware which might serve as some inspiration: https://docs.platformio.org/en/latest/plus/unit-testing.html

HotelCalifornia, 2019-09-20:
Sure we could add a manually run (presumably still automated) verification step that runs on a local V5, but I still think our first step should be something that can run in our already present azure pipeline

Octogonapus, 2019-09-20:
I agree. You might have missed my implication, though, that with PIO's approach, system tests that run on a local V5 can also be run through Azure.

HotelCalifornia, 2019-09-20:
I know about local agents, but I'm not sure how we can add one to our workflow easily.

WillXuCodes, 2023-01-12:
For future generations/devs who work on this, here is the place you need to add new unit tests:

https://github.com/purduesigbots/pros/tree/develop/src/tests

The reason I say this is because we've added a lot more sensors, but haven't been adding tests for them. Please do this when you add new sensors from now on. 

Furthermore, we need somebody to go in and add those new sensors so this is def a task that needs to be done.

PixelToast, 2023-01-26:
🥺 👉 👈