purduesigbots/pros #32: Ultrasonic sensor seems to return zero periodically

bradamiller, 2017-07-27:
#### Expected Behavior:
The VEX Ultrasonic sensor should always return a reasonable value for the distance
#### Actual Behavior:
I'm seeing a zero value about every 50ms in between the correct values.
#### Steps to reproduce:
I wrote this test program that gets the time of  a zero value and subtracts the last time we saw a zero
to look for a repeatable period:
```
 Ultrasonic ultra;
 const int leftMotor = 1;
 const int rightMotor = 10;

void operatorControl() {
		ultra = ultrasonicInit(4, 3);
		int lastTime = 0;
		while (1) {
			int distance = ultrasonicGet(ultra);
			if (distance == 0) {
				// if this is a zero range value
				unsigned long thisTime = millis();
				int time = thisTime - lastTime;  // time since last zero
				printf("millis: %d\n", time);
				while (ultrasonicGet(ultra) == 0); // wait for the distance to become nonzero
				lastTime = thisTime; // update the last time we saw a zero
			}
		}
}
```
#### System information:
mac os 10.12.6

Kernel Version:
2.12

#### Additional Information

HotelCalifornia, 2017-07-27:
Have you tried running this test with multiple ultrasonic sensors?

In our experience, these sensors have been the most troublesome to get working on the kernel side of things, so our first instinct is to chalk this up as a hardware issue.

bradamiller, 2017-07-27:
I had some students report that they had to exclude zero values so I tried it myself with a different robot and ended up writing that test program. I’ve seen those sensor return outlier values occasionally with other libraries, but the consistent 50ms thing made me think of some kind of race condition in the code.  I’ll try a few other sensors and see if the results are the same or it is a bad sensor.

I wrote the easyC library for the PIC vex controller and I remember those sensors being pretty problematic.

stcarlso, 2017-07-28:
https://github.com/purduesigbots/pros/blob/master/src/ultrasonic.c#L39 sets the value to zero when the sensor starts pinging, and https://github.com/purduesigbots/pros/blob/master/src/ultrasonic.c#L53 sets the correct value when the ping returns. Since the ping is broadcast every 50ms, checking the value repeatedly would be expected to return zero while the ping is outbound and the sensor waiting for the reply.

Commenting out line 39 will introduce issues with the old value being retained if the sensor detects nothing. It may be possible to only zero the value if the sensor times out (perhaps before line 65), but multiple sensor installations will then suffer as ping crosstalk from one ultrasonic sensor will randomly reset the previously correct values of other sensors.

Your best shot is probably to slow down the check for the sensors to at most once per millisecond (which will also prevent task starvation in the future) and to sanity check the values for zero.

HotelCalifornia, 2017-08-09:
I feel as though the problem here is that we are using the zero value both to mean that the sensor detects something at "zero" distance and as a default value for when the sensor is pinging (à la `undefined` or `NULL`).

> Shouldn’t the sensor value always be the previous value until it’s replaced with a new real value rather than ever returning the zero?

I worry that this might be assuming too much. Another solution would be to set the value to a constant, something like `#define ULTRASONIC_UNDEF 99999`

bradamiller, 2017-08-09:
The undefined value makes sense to handle the timeout case. But I believe that it should not ever return it when the sensor is getting valid data. The robots don’t move that fast and just returning the previous good value makes the programming easy for students and avoids the issue of them having some loop time where they never see an actual measurement because they coincidentally only read it while it’s waiting for the echo return.

baylessj, 2017-08-12:
I tested the ultrasonic driver both with and without line 39 and it looks like there are pros and cons to both options. I found the issue of the false zero every 50ms with line 39 present and also found in this configuration that there were a number of zeros reported when the ultrasonic was measuring near the edge of its range, but there would not be any zeros reported when line 39 was removed. I'm fairly confident that these zeros were the result of the ping timing out, so the zero value would be a clue to the user that the data being returned is no longer valid. The configuration without line 39 would lead the user to believe that they were still just shy of the max range, not actually exceeding it. I am typically in support of doing whatever we can to make PROS easier for students to use, but I'm wary of the potential for sacrificing data by removing line 39. You can always filter out data but you can't get more data than the driver provides, so I would be in favor of maintaining the current status of the driver and encouraging users to filter out the 50ms zeros when not near the top of the ultrasonic's range.

FWIW, I polled the ultrasonic every 20ms in the main op control loop. I saw some 50ms zeros this way, but I primarily got correct data. I can't think of any significant reasons for polling them at a non-standard interval, so my recommendation would be to maintain a 20ms loop even if the ultrasonics are being read in a different task.

HotelCalifornia, 2017-08-12:
> I'm fairly confident that these zeros were the result of the ping timing out, so the zero value would be a clue to the user that the data being returned is no longer valid.

> ... and encouraging users to filter out the 50ms zeros when not near the top of the ultrasonic's range.

I think this supports the idea that we add a defined value for the result of a ping timeout, instead of keeping those as zero and leaving users to guess what the zero means.

bradamiller, 2017-08-12:
I totally agree with you. If it only returns valid values or ping timeouts then it would be pretty simple to use. And if the timeout value was just some large range value then proportional controllers for driving to some distance would just work - the occasional timeout would just look like a big distance causing a large error and a fast speed, which in most cases would be the correct behavior.

baylessj, 2017-08-12:
`ultrasonicGet()` returns a signed int, so we could define the timeout code to be -1. I would expect that if a user finds this aspect of the sensor to be an issue then they would be viewing the data through the terminal output, so the timeout code will need to be rather noticeable (either negative or very large) so it can easily be identified as a non-standard value. Defining the error code as a negative value would be more noticeable in the aformentioned proportional controller example, but that could potentially be more helpful in debugging the system. Setting the timeout code to slightly above the ultrasonic's range might allow the controller to run as expected without a filter, but the average user would likely never identify it as an issue if they desired to properly filter the data.

However, setting a larger timeout code might not be less noticeable in all circumstances though, because another common utilization of the ultrasonic sensors is as an exit condition for a movement, such as moving until the robot is x number of inches away, and in that case, the 50ms timeouts would cause the movement to end prematurely. Regardless of what we set the timeout code to, there is the possibility that it will cause some kind of implementation of the sensor to fail.

I believe that the best rationale for determining the timeout code should be ensuring that when the ultrasonic sensor returns an unexpected or "incorrect" value, it is as noticeable to the user as possible to save them debugging time. Oftentimes a system that fails catastrophically is better than one that fails silently.

HotelCalifornia, 2017-10-14:
#34

closed 2017-10-14 by HotelCalifornia