purduesigbots/pros #38: Arch Linux Install Script Dependency Missing

Willv19, 2017-10-21:
I installed pros while running arch linux and attempted to build a project, but the compiler failed because it was missing an stdlib.h file. After doing some research, installing the "arm-none-eabi-newlib" fixes the issue because that is the package that provides the arm standard library. That package is an optional dependency of the arm gcc toolkit, but is not automatically installed by the install script. It should be added for the next release.

HotelCalifornia, 2017-10-23:
The Arch Linux install script that we packaged with some of our older releases was provided by a member of the community, so we make no guarantees as to its effectiveness. That being said, if you have a version of the script that you feel improves upon the old one, feel free to post it here and we can add it to our releases.

Willv19, 2017-10-26:
How should I post a new version of the script? Github gist?

HotelCalifornia, 2017-10-26:
Sure, or just leave it in a comment here

Willv19, 2017-10-26:
Here it is: https://gist.github.com/Willv19/8a64ec1e176ef7227fe905a47f4cd08e. It is essentially the same as the old install script but it installs the extra required dependency as well

closed 2018-12-20 by HotelCalifornia