purduesigbots/pros #39: Atom - can't create a new project

ben06202001, 2017-10-23:
#### Expected Behavior:
I tried creating a new project in Atom (I have atom, python 3.6, and have downloaded and installed the macos package for pros) I dont think I have clang but once I downloaded it I was pretty much lost (what do I do now?). 
#### Actual Behavior:
no new project, the error says that the issue has already been reported https://github.com/purduesigbots/pros-atom/issues/37
but this doesn't seem to be the issue
this seems closer
https://github.com/purduesigbots/pros-atom/issues/6
#### Steps to reproduce:
//
#### System information:
Kernel Version: none
Here is what I think (dont quote be though) seems to be the problem. I have no kernels installed and cannot install any.

I have tried 'pros conduct download' but the responce was:

Fetching online listing to verify available templates.
No templates were found with the name kernel
Aborted!

I am not very familiar with all the terminology that is used and how most things work so I probably did something wrong. But I am lost. 

#### Additional Information

Thank you in advance and sorry

edjubuh, 2017-10-23:
Have you run `pros conduct first-run` from the terminal?

HotelCalifornia, 2017-10-23:
Hi, can I ask how you went about installing PROS?

ben06202001, 2017-10-23:
1) No I havent ran the pros conduct first-run form terminal, I didn't know, I just ran it but after being proposed to 'Download the latest kernel' still the answer was 'No templates were found with the name kernel'

2) I installed PROS many times, I use to have atoms and python 2.6, I downloaded PROS but got lost and uninstalled everything. I did a bit of digging and reinstalled Atom, Python 3.6, and used the macOS package to install PROS. I did get an error at the end of the installation saying that the installation couldn't be completed. Still PROS was there and everything seemed fine. (In the first runs though I did get the PROS CLI as a separate aplication but couldn't open it as it required an os further than my current one, that seemed odd as I did not know of one). 

Thank you for replying so fast and for your help,

HotelCalifornia, 2017-10-23:
`PROS CLI.app` is not really meant to be opened, but run from the terminal, so that particular error can be ignored.

HotelCalifornia, 2017-10-23:
That being said, the underlying cause of this issue is due to a mistake in the build environment for the CLI on MacOS. Please see purduesigbots/pros-cli#45 for more on this issue

closed 2017-10-23 by HotelCalifornia