purduesigbots/pros #41: timer

3038922, 2017-11-28:
#### Expected Behavior:

#### Actual Behavior:

#### Steps to reproduce:

#### System information:
Kernel Version:

#### Additional Information

3038922, 2017-11-28:
I am a senior high school student. Before I used EASYC, I would like to ask how to write timer functions. Just like EASYC, there are starttimer (1); presstimer (1,0); gettimer (1); stoptimer (1); Thank you.

edjubuh, 2017-11-28:
I haven't used EasyC in many years and I'm not familiar with those functions. What are you trying to accomplish?

3038922, 2017-11-28:
void pid_encoder_FR ( long target, long tolerance, unsigned long time )
{
    // 前进为+ 后退为-
    static int lasterror = 0;
    static int derivative = 0;
    static int integeral = 0;
    static int error = 0;
    static int insidetarget = 0;
    if ( fr.control == OFF && fr.ready == OFF )
    {
        PresetTimer ( 3 , 0 ) ;
        PresetTimer ( 4 , 0 ) ;
        lasterror = 0;
        derivative = 0;
        integeral = 0;
        error = 0;
        insidetarget = 0 ;
        PresetQuadEncoder ( 3 , 4 , 0) ;
        fr.control = ON ;
    }
    else if ( fr.control == ON )
    {
        if ( (insidetarget == 0 || GetTimer ( 3 ) < 200 ) && GetTimer ( 4 )  <= time )
        {
            error = target - GetQuadEncoder ( 3 , 4 ) ;
            if ( abs(error) <= tolerance )
            {
                insidetarget = 1 ;
            }
            else
            {
                insidetarget = 0 ;
                PresetTimer ( 3 , 0 ) ;
            }
            if ( abs(error) < 50 && (error != 0) )
            {
                integeral = integeral + error ;
            }
            else
            {
                integeral = 0 ;
            }
            derivative = error - lasterror  ;
            lasterror = error ;
            fr.pwm[0] = (int)((float)error * fr.kp) + ((float)integeral * fr.ki) + ((float)derivative * fr.kd);
            //PrintToScreen ( "pid_encoder_motor.pwm[0]:%d\n" , fr.pwm[0] ) ;
            if ( fr.pwm[0] > 127 )
            {
                fr.pwm[0] = 127 ;
            }
            if ( fr.pwm[0] < -127 )
            {
                fr.pwm[0] = -127 ;
            }
            motor_drive_FR ( fr.pwm[0] ) ;
            Wait ( 5 ) ;
        }
        else
        {
            if ( PRINT_TO_SCREEN == ON )
            {
                PrintToScreen ( "pid_encoder_FR:%d\n" , GetQuadEncoder ( 3 , 4 ) ) ;
                PrintToScreen ( "pid_encoder_FR_time:%d\n" , GetTimer ( 4 )  ) ;
            }
            motor_drive_FR ( 0 ) ;
            lasterror = 0;
            derivative = 0;
            integeral = 0;
            error = 0;
            insidetarget = 0 ;
            fr.control = OFF ;
            fr.ready = ON ;
        }
    }
}

3038922, 2017-11-28:
Many of my functions are based on a number of timers.

3038922, 2017-11-28:
I want to change my EASYC program into a pros program

3038922, 2017-11-28:
void time_intake_control ( int x, unsigned long y )
{
    if ( intake.control == OFF && intake.ready == OFF )
    {
        PresetTimer ( 2 , 0 ) ;
        intake.control  = ON ;
    }
    else if ( intake.control == ON )
    {
        if ( GetTimer ( 2 ) <= y )
        {
            motor_intake_cone ( x ) ;
            intake.ready = OFF ;
        }
        else
        {
            motor_intake_cone ( 0 ) ;
            intake.control  = OFF ;
            intake.ready = ON ;
        }
    }
}

edjubuh, 2017-11-28:
PROS doesn't have any identical functions, but you might be interested using the `millis()` function which returns the number of milliseconds the robot has been on.

```
unsigned long start_timer = millis(); // equivalent of PresetTimer to 0
// do stuff
unsigned long timer_val = millis() - start_timer; // equivalent of GetTimer
if(timer_val < 200) {
   // do stuff
}
```

As a side note, I'd recommend thinking about what your code does as a whole, and then porting that as opposed to doing a line by line conversion from EasyC to PROS.

3038922, 2017-11-28:
3qqq

EricHallahan, 2017-11-28:
Consider this closed.

closed 2017-11-28 by HotelCalifornia