purduesigbots/pros #42: Windows Toolchain uses 64-bit binaries

edjubuh, 2017-12-22:
Toolchain works for 32-bit and 64-bit systems. The new toolchain accidentally snuck in 64-bit binaries, so need to rebuild the toolchain.

Reported from a PROS Development email.

Current workaround is to install old toolchain (gcc-toolchain-win-old.zip from the 2.11.1 release) and update C:\Program Files\PROS\toolchain\usr appropriately.

3038922, 2017-12-22:
My lint on-the-fly function cannot be used.
My desktop computer can be used.
[image]

edjubuh, 2017-12-22:
Hi @3038922, can you create a new issue for this on [purduesigbots/pros-atom](https://github.com/purduesigbots/pros-atom)?

3038922, 2017-12-22:
OK.sorry

closed 2019-01-30 by edjubuh