purduesigbots/pros #45: Task Deletes

jcgrif, 2018-02-20:
**Expected Behavior:**  When deleting tasks we would expect them be removed and resources freed.

**Actual Behavior:**  When we delete tasks, the task counts never drop.  If we create a task, let it do its job and then have it call taskDelete(NULL) or by using the taskhandle, the task count never drops.

**Steps to reproduce:**  In addition to the standard task, there are 4 other tasks that are running plus the one that we are trying to use to create and delete itself.  The task count increase until it reaches 17 and then fails because the task limit is set to 16.

#### System information:
Kernel Version: 2.6.1

#### Additional Information

jabibi100, 2018-02-20:
@jcgrif can you provide a sample project/code where you observe this behavior. I believe I know the root cause of the issue you are observing but I want to confirm.

jabibi100, 2018-03-21:
@jcgrif any updates on this?

HotelCalifornia, 2018-07-06:
closing due to inactivity

closed 2018-07-06 by HotelCalifornia

jcgrif, 2018-07-09:
I will test on the V5 once it hits a released version.  Haven't had an opportunity with the Cortex to try to reproduce it.