purduesigbots/pros #48: C++ Global Constructor Never Called

sjsch, 2018-02-23:
#### Expected Behavior:

When a C++ object is declared to be global, rather than inside a function, its constructor is not called.

#### Actual Behavior:

The object remains uninitialized.

#### Steps to reproduce:

Using the default template, replace opcontrol.c with opcontrol.cpp:

```
#include "main.h"

class test {
public:
	test();
	int y;
};

test::test()
{
	y = 3;
	printf("Test constructor\n");
}

test x;

void operatorControl() {
	printf("opcontrol\n");
	printf("test: %d\n", x.y);

	while (1) {
		delay(20);
	}
}
```

#### System information:
Kernel Version: 2.11.1

PixelToast, 2018-02-23:
1. Upgrade your kernel
2. C++ Isn't officially supported by PROS

Static constructors arent called because `gcc`/`ld` are doing all the linking instead of g++ and doesn't automatically call `__libc_init_array` the solution is to call it manually, see: https://lab.pxtst.com/PixelToast/8762A-2016/blob/master/src/init.cpp

sjsch, 2018-02-24:
Thanks for the worked example!  I knew it had something to do with that, but I was having trouble tracking down the solution.  I'll close this now, because it isn't a PROS issue.

closed 2018-02-24 by sjsch

CanyonTurtle, 2018-07-10:
The fix provided by @PixelToast causes a segmentation fault for me, on kernel version 2.12.1. Do you have any recommendations @PixelToast for getting static constructors to behave properly in PROS 2.X.X C++?