purduesigbots/pros #49: Remote Control Status Light Blinking Red: Fault: User Microprocessor Issue

kfish610, 2018-03-11:
On Windows, when downloading to the robot, 1 in 5 or 10 times I will get: 
```
Asking for system information... Asking for system information... Failed to get system info... Prompting to continue.
..
Cortex Microcontroller connected on
  Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W  w/ 0.00V
  Cortex: F/W  w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. Continue?
```
And then the robot light becomes blinking red, indication fault: User Microprocessor Issue
This can be temporarily fixed by either downloading over USB instead of VEXNet or reflashing the firmware, but this always only fixes it until a few more times. Why is this?

edjubuh, 2018-03-12:
What version of the PROS CLI are you using?

How are you flashing code? You can run `pros upload --debug` to gain some additional debug output which will be useful for us to help diagnose what's going on.

kfish610, 2018-03-13:
PROS CLI: 2.6.1 - Package: 0.7.4
Flashing Over VexNet.

kfish610, 2018-03-14:
Debug output:
```
  ====:: PROS Flasher v2.6.1 ::====
        DEBUG: Opening /mnt/c/Users/kfish/Documents/GitHub/CHSVexRoboticsInTheZonePROS8771Y/project.pros (ProjectConfig)
Flashing /mnt/c/Users/kfish/Documents/GitHub/CHSVexRoboticsInTheZonePROS8771Y/bin/output.bin to /dev/ttyS3
Stopping user code...   DEBUG: 0F 0F 21 DE 08 00 00 00 08 F1 04
        DEBUG:
complete
Asking for system information...        DEBUG: SYS INFO BITS: C9 36 B8 47 21   PORT CFG: Serial<id=0x7f9d10414fd0, open=True>(port='/dev/ttyS3', baudrate=115200, bytesize=8, parity='E', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
        DEBUG: SYS INFO RESPONSE: AA 55 21 0A 04 19 04 19 80 82 01 04 00 04
complete
Cortex Microcontroller connected on /dev/ttyS3
  Tether:   Serial w/ VEXnet 2.0 Keys
Joystick: F/W 4.25 w/ 7.55V
  Cortex: F/W 4.25 w/ 7.67V (Backup: 0.00V)
Sending to download channel (this may take a while)... complete
Exposing bootloader... complete
Preparing bootloader...         DEBUG: STM BL RESPONSE TO 0x7F:
        DEBUG: STM BL RESPONSE TO 0x7F:
        DEBUG: STM BL RESPONSE TO 0x7F:
failed
Retrying...
Stopping user code...   DEBUG: 0F 0F 21 DE 08 00 00 00 08 F1 04
        DEBUG:
complete
Asking for system information...        DEBUG: SYS INFO BITS: C9 36 B8 47 21   PORT CFG: Serial<id=0x7f9d10414d68, open=True>(port='/dev/ttyS3', baudrate=115200, bytesize=8, parity='E', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
Asking for system information... Failed to get system info... Prompting to continue...
Cortex Microcontroller connected on
  Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W  w/ 0.00V
  Cortex: F/W  w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. Continue? [Y/n]: Y
Exposing bootloader... complete
Preparing bootloader...         DEBUG: STM BL RESPONSE TO 0x7F:
        DEBUG: STM BL RESPONSE TO 0x7F:
        DEBUG: STM BL RESPONSE TO 0x7F:
failed
Retrying...
Stopping user code...   DEBUG: 0F 0F 21 DE 08 00 00 00 08 F1 04
        DEBUG:
complete
Asking for system information...        DEBUG: SYS INFO BITS: C9 36 B8 47 21   PORT CFG: Serial<id=0x7f9d10414e80, open=True>(port='/dev/ttyS3', baudrate=115200, bytesize=8, parity='E', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
Asking for system information... Failed to get system info... Prompting to continue...
Cortex Microcontroller connected on
  Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W  w/ 0.00V
  Cortex: F/W  w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. Continue? [Y/n]: n
Aborted!
`

kfish610, 2018-03-16:
A little more:
```
 ====:: PROS Flasher v2.6.1 ::====
        DEBUG: Opening /mnt/c/Users/kfish/Documents/GitHub/CHSVexRoboticsInTheZonePROS8771Y/project.pros (ProjectConfig)
Flashing /mnt/c/Users/kfish/Documents/GitHub/CHSVexRoboticsInTheZonePROS8771Y/bin/output.bin to /dev/ttyS3
Stopping user code...   DEBUG: 0F 0F 21 DE 08 00 00 00 08 F1 04
        DEBUG:
complete
Asking for system information...        DEBUG: SYS INFO BITS: C9 36 B8 47 21   PORT CFG: Serial<id=0x7f7288d14c88, open=True>(port='/dev/ttyS3', baudrate=115200, bytesize=8, parity='E', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
        DEBUG: SYS INFO RESPONSE: 00 00 AA 55 21 0A 04 19 04 19 7A 7A 00 04 00 00
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
Asking for system information... complete
Cortex Microcontroller connected on /dev/ttyS3
  Tether:   Serial w/ VEXnet 2.0 Keys
Joystick: F/W 4.25 w/ 7.20V
  Cortex: F/W 4.25 w/ 7.20V (Backup: 0.00V)
Sending to download channel (this may take a while)... complete
Exposing bootloader... complete
Preparing bootloader...         DEBUG: STM BL RESPONSE TO 0x7F:
        DEBUG: STM BL RESPONSE TO 0x7F:
        DEBUG: STM BL RESPONSE TO 0x7F:
failed
Retrying...
Stopping user code...   DEBUG: 0F 0F 21 DE 08 00 00 00 08 F1 04
        DEBUG:
complete
Asking for system information...        DEBUG: SYS INFO BITS: C9 36 B8 47 21   PORT CFG: Serial<id=0x7f7288d14d68, open=True>(port='/dev/ttyS3', baudrate=115200, bytesize=8, parity='E', stopbits=1, timeout=0.5, xonxoff=False, rtscts=False, dsrdtr=False)
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
        DEBUG: SYS INFO RESPONSE:
Asking for system information... Failed to get system info... Prompting to continue...
Cortex Microcontroller connected on
  Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W  w/ 0.00V
  Cortex: F/W  w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. Continue? [Y/n]: n
`