purduesigbots/pros #50: system("cls") can't use.

3038922, 2018-03-14:
I want to write an online monitoring. I tried the system ("CLS") command, and I couldn't clear the command line. I want to know what the function of the console is to clear the command line.

HotelCalifornia, 2018-03-14:
For clearing the screen, try using [ASCII escape codes](), like:

```c
// clear screen and reset cursor to top left
printf("\033[2J");
```
or
```c
// clear screen, resetting cursor to top left and deleting scrollback buffer
printf("\033[3J");
```

3038922, 2018-03-15:
Is there a function like scanf to receive keyboard information in PORS?

I enter a string, such as AUTO INIT., and then execute the related function.

jcgrif, 2018-03-15:
You can use fgetS(buffer, maxcharacters, stdin) to get a string

HotelCalifornia, 2018-03-15:
As for your other question about LLVM, I'm not sure why you would need anything else. We only use LLVM to provide static analysis and autocompletion in Atom. I'm going to close this, but feel free to reopen it if you run into problems with the solutions posted here.

closed 2018-03-15 by HotelCalifornia