purduesigbots/pros #51: How to update the firmware of the master controller and the remote control?

3038922, 2018-03-17:
EASYC is the main controller and the remote controller need to brush special firmware. Does PROS need? I doubt that my TASK is always the data error, because no update to the latest version of the firmware. The key to my remote or in the use of EASYC firmware.

jabibi100, 2018-03-21:
Hey @3038922 PROS does not require any special firmware to run on the Vex Cortex. When you click download inside of Atom or use the CLI to flash it uploads all necessary software to the cortex for it to run.

However, if you are asking about the Vex master firmware, controller/joystick firmware, or Vexnet 2.0 keys you can find that information here [https://www.vexrobotics.com/vexedr/resources/firmware](https://www.vexrobotics.com/vexedr/resources/firmware). We do recommend using the latest firmware from Vex at all times. Just be sure to re-upload your program after you have flashed the master firmware to the cortex.

3038922, 2018-03-21:
Thank you. I am a beginner. There are always many problems. My country can't open YOUTUBE.. My English is not good enough. So I can't learn PROS. well enough.

jabibi100, 2018-03-22:
Not a problem at all. I am going to close this issue now.  Additionally, feel free to ping me on the [Vex discord server ](https://discordapp.com/channels/197777408198180864/198658294007463936) as well for any future small issues so that you don't have to open Github issues.

If you don't mind me asking which country do you reside? We have a very diverse team and may have a team member who could speak in your native tongue for future issues.

closed 2018-03-22 by jabibi100

3038922, 2018-03-23:
I come from china.For some strange reasons, China can't visit GOOGLE YOUTUBE FACEBOOK., which makes it difficult for us to learn.My Team Number is 8982 7995 7671.We are the 2015 2016 Chinese Champions .
 We are good at operating and mechanical structures. We are not good at programming. We have been using EASYC. all the time.
Thack you.

3038922, 2018-03-23:
We are learning the class of C++. We don't have enough experience in designing architecture. I don't know how to design the overall architecture to make the code look more graceful.