purduesigbots/pros #53: Encoder data error!

3038922, 2018-03-20:
[image]
[image]
[image]
[image]

I think my code is the right line and it's not plugged. But encoder_R's data is two times that of encoder_L.encoder_L's data is right.

edjubuh, 2018-03-20:
Ensure that the encoder is in good electrical condition - verify the wires are plugged in and that there is no damage to the encoder. Closing this issue since there doesn't seem to be an issue with PROS.

closed 2018-03-20 by edjubuh

3038922, 2018-03-21:
I'm sure the installation is correct, because it's normal under EASYC. We are all eight robots in our team.

3038922, 2018-03-21:
Will it be the reason that I use C++11?

jbuschjr, 2018-03-21:
Does the issue still occur if you use different digital ports for encoder_R?  Such as
`encoder_R = encoderInit(5, 6, false);`

3038922, 2018-03-21:
5,6 can. It seems that the tutorial shows that 10 11 12 ports can't be used in the case of C++11.
Thank you veay mush!