purduesigbots/pros #54: Is my C++ setup wrong?

3038922, 2018-03-26:
[image]
I only modified this, and then changed the.C to.Cpp., it could be compiled and passed.
[image]
I see that in the VEX forum, the setting of C++ is quite complex.
The constructor and the destructor are invalid. But it doesn't matter, I initialized it by myself.
My biggest problem now seems to be that when I connect to the field control, after the automatic race is over, the manual mode needs to wait for a second to operate the remote controller. It doesn't matter if we operate the remote controller. If we don't wait, the data generated by TASK1 and TASK2 will go wrong.

3038922, 2018-03-26:
/**
 * @Author: 陈昱安
 * @Date:   2018-01-03T14:49:32+08:00
 * @Filename: init.c
 * @Last modified by:   陈昱安
 * @Last modified time: 2018-01-03T22:53:46+08:00
 */
#include "main.h"
void initializeIO()
{
    watchdogInit(); //静电冲击重置模块
}
void initialize()
{

    printf("mainbattery:%.3f\n", (float)powerLevelMain() / 1000.0);
    printf("Power_Expander_battery:%.3f\n", (float)analogRead(8) / Power_Expander / 4.0);
    printf("Backupbattery:%.3f\n", (float)powerLevelBackup() / 1000.0);
    Init_system();
    Init_fr();
    Init_rotate();
    Init_spinner();
    Init_lift();
    Init_intake();
    Init_MBC();
    auto_Init();                  //初始化
    spinnerMutex = mutexCreate(); //先打开互斥锁
    liftMutex = mutexCreate();
    encoder_gyroMutex = mutexCreate();
    task1 = taskRunLoop(taskPoten, 20); //再创建TASK
    task2 = taskRunLoop(taskEncoderGyro, 20);
    if (!digitalRead(7)) // 蓝方装填区&&红方野区
        print("Autonomous_zone:BLUE_preload&&RED_outsid\n");

    else if (!digitalRead(8)) //红方野区&&蓝方装填区
        print("Autonomous_zone:RED_preload&&BLUE_outsid\n");
    else
        print("WARNING!!!WARNING!!! Please insert jumper to 7 or 8 !!\n");
    if (!digitalRead(9)) // 20分区
        print("Autonomous:20point\n");
    else if (!digitalRead(10)) // 10分区
        print("Autonomous:10point\n");
    else
        print("WARNING!!! WARNING!!!  Please insert jumper to 9 or 10 !!\n");
    if (!digitalRead(11)) //两个个帽子
        print("Autonomous:2 cone\n");
    else if (!digitalRead(12)) // 三个帽子
        print("Autonomous:3 cone\n");
    else
        print("Autonomous:1 cone\n");
    if (DEBUG_MODE)
        print("debug_open\n");
    else
        print("debug_close\n");
    printf("Version : %s\n", VERSION);
}
void Init_system()
{
    gyro = gyroInit(GYRO_PORT, ROTATE_ERROR_CORRECTION); //陀螺仪初始化 //默认值 196
    encoder_L = encoderInit(3, 4, false);                //左边编码器初始化
    encoder_R = encoderInit(5, 6, false);                //右边编码器初始化
    sys.m_nStep = 1;
    sys.m_nPlayer = 1;
    sys.m_nCone = 0;
    sys.m_isPreload = false;
    sys.m_isAutoMode = false;
    sys.m_isJoystick_mode = false;
    //watch_Dog 设置
    sys.m_isDtFlag = false;
    sys.m_uDtSet = 20; //系统10MS循环一次
[... 87 lines cut ...]
    {CONE9_SPINNER_VALUE, CONE9_LIFT_VALUE, CONE9_PRELOAD_INTAKE_TIME, CONE9_LIFT_DOWN_TIME, CONE9_LIFT_UP_TIME},
    {CONE10_SPINNER_VALUE, CONE10_LIFT_VALUE, CONE10_PRELOAD_INTAKE_TIME, CONE10_LIFT_DOWN_TIME, CONE10_LIFT_UP_TIME},
    {CONE11_SPINNER_VALUE, CONE11_LIFT_VALUE, CONE11_PRELOAD_INTAKE_TIME, CONE11_LIFT_DOWN_TIME, CONE11_LIFT_UP_TIME},
    {CONE12_SPINNER_VALUE, CONE12_LIFT_VALUE, CONE12_PRELOAD_INTAKE_TIME, CONE12_LIFT_DOWN_TIME, CONE12_LIFT_UP_TIME},
    {CONE13_SPINNER_VALUE, CONE13_LIFT_VALUE, CONE13_PRELOAD_INTAKE_TIME, CONE13_LIFT_DOWN_TIME, CONE13_LIFT_UP_TIME},
    {CONE14_SPINNER_VALUE, CONE14_LIFT_VALUE, CONE14_PRELOAD_INTAKE_TIME, CONE14_LIFT_DOWN_TIME, CONE14_LIFT_UP_TIME},
    {CONE15_SPINNER_VALUE, CONE15_LIFT_VALUE, CONE15_PRELOAD_INTAKE_TIME, CONE15_LIFT_DOWN_TIME, CONE15_LIFT_UP_TIME},
    {CONE16_SPINNER_VALUE, CONE16_LIFT_VALUE, CONE16_PRELOAD_INTAKE_TIME, CONE16_LIFT_DOWN_TIME, CONE16_LIFT_UP_TIME},
    {CONE17_SPINNER_VALUE, CONE17_LIFT_VALUE, CONE17_PRELOAD_INTAKE_TIME, CONE17_LIFT_DOWN_TIME, CONE17_LIFT_UP_TIME}};
CINFO spinner; // spinner.control[1] spinner_tall_secretory //spinner.control[2]
CINFO lift;
CINFO MBC;
CINFO intake; // intake.time[0] preload_intake_time
CINFO fr;     // fr.time[0]为程序间隔dt fr.time[1]为存放时间
CINFO rotate;
// Debug debug[500];
SYSTEM sys;
Gyro gyro;
Encoder encoder_L;
Encoder encoder_R;
TaskHandle task1;        //线程1 FOR POTEN
TaskHandle task2;        //线程2 FOR ENCODER GYRO
Mutex spinnerMutex;      //互斥锁1
Mutex liftMutex;         //互斥锁2
Mutex encoder_gyroMutex; //互斥锁3

3038922, 2018-03-26:
auto.c  I haven't written anything yet

3038922, 2018-03-26:
/**
 * @Author: 陈昱安
 * @Date:   2018-01-03T14:49:32+08:00
 * @Filename: opcontrol.c
 * @Last modified by:   陈昱安
 * @Last modified time: 2018-01-04T00:31:04+08:00
 */

#include "main.h"
void operatorControl()
{

#if (DEBUG_MODE)
    startPage();
#else
    joystickControl();
#endif
}
void joystickControl()
{
    taskDelete(task2);
    mutexDelete(encoder_gyroMutex);
    gyroShutdown(gyro);
    stopEncoder();
    while (true)
    {
        sys.m_uTimeNow = millis();
        arcad4(joystickGetAnalog(1, 3), joystickGetAnalog(1, 1), 127);
        operator_auto_button();
        if (sys.m_isAutoMode == true)
        {
            auto_mode();
        }
        else
        {
            if (sys.m_nPlayer == 1)
            {
                if (sys.m_isJoystick_mode == 0)
                {
                    operator_lift();
                    operator_intake_cone();
                    motor_MBC(10);
                    if (spinner.m_isControl[1] == true)
                    {
                        if (spinner.m_isControl[2] == 1)
                            spinner.pidControl(cone_info[sys.m_nCone][0], 20, 700);
                        else
                            spinner.pidControl(SPINNER_VALUE_TALL_SECRETORY, 20, 700);
                    }

                    else
                        lifting_spinner_control();
                }
                else if (sys.m_isJoystick_mode == 1)
                {
                    operator_spinner();
                    operator_MBC_lift();
#if defined(NOSENSOR)
#else
                    if ((joystickGetDigital(sys.m_nPlayer, 6, JOY_UP) || (joystickGetDigital(sys.m_nPlayer, 6, JOY_DOWN))) && lift.m_nValueKalman < LIFT_VALUE_PRELOAD)
                    {
                        lift.p_motor(lift.pidCalc(CONE5_LIFT_VALUE, 30));
                    }
                    else
                    {
                        lifting_lift_control();
                    }
#endif
                    motor_intake(INTAKE_LIFTING_CLOSE);
                }
            }
            else if (sys.m_nPlayer == 2)
            {
                operator_intake_cone();
                operator_MBC_lift();
#if defined(NOSENSOR)
                operator_lift();
#else
                if ((joystickGetDigital(sys.m_nPlayer, 6, JOY_UP) || (joystickGetDigital(sys.m_nPlayer, 6, JOY_DOWN))) && lift.m_nValueKalman < LIFT_VALUE_PRELOAD)
                {
                    lift.p_motor(lift.pidCalc(CONE5_LIFT_VALUE, 30));
                }
                else
                {
                    operator_lift();
                }
#endif
                if (spinner.m_isControl[1] == true)
                {
                    if (spinner.m_isControl[2] == 1)

                        spinner.pidControl(cone_info[sys.m_nCone][0], 20, 700);
                    else
                        spinner.pidControl(SPINNER_VALUE_TALL_SECRETORY, 20, 700);
                }

                else
                {
                    operator_spinner();
                }
            }
        }
        waitUntil(&sys.m_uTimeNow, sys.m_uDtSet);
    }
}

jabibi100, 2018-04-05:
While it is theoretically possible, as PixelToast has shown in his blog, to use C++ with PROS on the Vex Cortex Microcontroller it is not officially supported by the dev team.

As a heads up issues opened on GitHub are treated as tasks that need to be performed. We close issues that are questions because there is no task to perform. For questions about programming with PROS, there are several options:

- [PROS Docs](https://pros.cs.purdue.edu/tutorials)
- [VEX Forum](https://www.vexforum.com/)
- [PROS Development Slack Team](https://join.slack.com/t/pros-development/shared_invite/enQtMzIyNzA2MTU2MDgwLWNlNTk3ZjA3ZWZmM2Q0YzI3NDZkZWE0MzU4ODdkM2U5MzhjZDY4M2RjNzBiZWYzODk0MTdkYzBiMzJjMWUyYmI)
- The #coding channel on the [Unofficial VEX Discord Server](https://discord.gg/32VrqVt)

The VEX Forum, the Slack team, and Discord all have very helpful community members that are happy to help with questions about using PROS.

closed 2018-04-05 by jabibi100