purduesigbots/pros #55: Please recommend a suitable Bluetooth chip.Thank you

3038922, 2018-04-05:
Direct use of printf always shows errors. I would like to invite you to recommend a Bluetooth chip that can be connected to UART port, so that I can make it easier when I debug. Thank you.

3038922, 2018-04-05:
https://www.vexforum.com/index.php/14783-bnsbluetooth-robotc-library-for-the-hc-05
Can PROS support this?

edjubuh, 2018-04-05:
To my knowledge this bluetooth chip just requires a UART driver, which the PROS kernel has. This means it's possible to use PROS with this Bluetooth chip.

Issues opened on GitHub are treated as tasks that need to be performed. We close issues that are questions because there is no task to perform. For questions about programming with PROS, there are several options:
- [PROS Docs](https://pros.cs.purdue.edu/tutorials)
- [VEX Forum](https://www.vexforum.com)
- [PROS Development Slack Team](https://join.slack.com/t/pros-development/shared_invite/enQtMzIyNzA2MTU2MDgwLWNlNTk3ZjA3ZWZmM2Q0YzI3NDZkZWE0MzU4ODdkM2U5MzhjZDY4M2RjNzBiZWYzODk0MTdkYzBiMzJjMWUyYmI)
- The #coding channel on the [Unofficial VEX Discord Server](https://discord.gg/32VrqVt)

The VEX Forum, the Slack team, and Discord all have very helpful community members that are happy to help with questions about using PROS.

closed 2018-04-05 by edjubuh

3038922, 2018-04-06:
The VEX forum blocked the IP from China, perhaps because Chinese students always expose the shady IFI China in the forum.

3038922, 2018-04-06:
I can't register vex forum at all. It's necessary to use non Chinese IP.