purduesigbots/pros #63: invalid desktop entry for PROS editor on Debian

gnations56, 2018-07-04:
#### Expected Behavior:
When clicking the menu entry in XFCE under programming, PROS Editor should open
#### Actual Behavior:
Debian reports that the desktop file entry is invalid
#### Steps to reproduce:
install pros-editor deb package and try to launch using menu in desktop environment
#### System information:
Kernel Version: Linux 4.9.0-6-amd64
Debian Stretch
Desktop Environment: XFCE
#### Additional Information
launching from command line using pros-editor does work

iuyte, 2018-07-04:
What are the contents of the pros-editor desktop file? (probably in /usr/share/applications IIRC)

gnations56, 2018-07-04:
[Desktop Entry]
Name=PROS Editor
Comment=A hackable text editor for the 21st Century.
GenericName=Text Editor
Exec=/usr/share/pros-editor/atom %F
Icon=pros-editor
Type=Application
StartupNotify=true
Categories=GNOME;GTK;Utility;TextEditor;Development;
MimeType=text/plain;

iuyte, 2018-07-05:
What is the result of `which pros-editor` in the command line?

gnations56, 2018-07-09:
/usr/bin/pros-editor

iuyte, 2018-07-10:
A workaround would be to try changing the following line:
```
Exec=/usr/share/pros-editor/atom %F
```
to this:
```
 Exec=/usr/bin/pros-editor %F
```

gnations56, 2018-07-10:
Ok