purduesigbots/pros #64: Add Documentation about C++ with 2.X.X

CanyonTurtle, 2018-07-16:
C++ with PROS V2.X.X is unofficially supported, and that is fine. But I think that a little documentation could go a long way for people trying to get C++ to work on the VEX Cortex.

For instance, static constructors are not called for some reason(#48 provides a solution, but this has not worked for me.) This pitfall can be avoided with a documented warning about static constructors. I think the section of the `README.md` titled "Does PROS support C++?" should contain a blurb about this, such as:

One known issue with C++ projects is that static constructors are never called, which means objects defined in the global scope will not behave as expected. To avoid this problem, avoid putting objects and structs in the global scope. Instead, define global pointers to objects, and use dynamic memory allocation (via the `new` keyword, or via `malloc`) inside functions.

Some different ideas about how to incorporate this warning:
- put a comment directly in the user-generated templates that warns about this
- add a file in this repository to track C++ integration, such as `CPP-GUIDELINES.md`
- add to the website documentation about C++, and/or link to previously mentioned guideline file

I would be happy to make a PR either here or on the docs, if the maintainers would be supportive of such documentation.

HotelCalifornia, 2018-07-16:
From the README:

> No, PROS 2.x does not officially support C++. Some users have found ways around this, but be warned: **we will not be able to help if you run into issues doing this.**
(emphasis added)

It is not productive to add documentation relating to features we do not officially support, as doing so may in fact cause more confusion than it would clear up.

In any case, we appreciate your willingness to contribute.

closed 2018-07-16 by HotelCalifornia

CanyonTurtle, 2018-07-16:
I understand that adding docs for features that PROS is unwilling to maintain may be counterproductive. However, this specific problem is easy to run into, rather difficult to isolate, and easily addressed. At least I have put this information out there for anyone who looks by opening this issue.

Thanks for reading @HotelCalifornia !