purduesigbots/pros #72: UART/I2C/SPI over ADI?

sealj553, 2018-09-28:
Is there currently any way to connect either UART, I2C, or SPI devices over the ADI ports on the V5? Would bit-banged implementations be desired? 
Or would this kind of thing be better suited for OkapiLib?

edjubuh, 2018-09-28:
None of those protocols are possible with the ADI due to VEXos limitations. It is possible to implement a device which communicates over the smart ports using PROS. We're slowly working on documentation and a reference implementation for this.

I'm closing the issue since there is no work for us to do, but feel free to continue the discussion if you have questions about generic smart devices.

closed 2018-09-28 by edjubuh

sealj553, 2018-10-01:
Couldn't `adi_digital_read` and `adi_digital_write` be used to implement any arbitrary protocol in software?
Also, is there any existing documentation about VEXos and its limitations?

I see on the [V5 system architecture](https://www.vexrobotics.com/v5-architecture.html) page, it says:
"A dedicated Cortex M0 microcontroller has the sole responsibility of reading inputs and toggling outputs." and "Future enhancements to the 3-Wire ports are planned to allow I2C and UART communication to non-VEX sensors as well." 

Do we not have direct access to the microcontroller? I figure the enhancements they are talking about must be software-side.

Octogonapus, 2018-10-01:
Yes, those two functions could be used for that, but it would be extraordinarily slow. You only have access to high-level communications to the uC running the ADI ports.