purduesigbots/pros #78: Getting PROS to run on Ubuntu 14.04

norwalk350, 2018-10-29:
#### Expected Behavior:

We're trying to run PROS on our student 1:1 computers (primarily, for the robotics team). We're having issues getting PROS to communicate with the VEX Cortex Microcontroller (Logic PCB). I referenced the documentation here:

https://pros.cs.purdue.edu/cortex/getting-started/debian-linux.html

For our students, we run a custom Linux distro based off of Ubuntu 14.04 (Cub Linux). 

I know the documentation recommends to run Ubuntu 16.04 (particularly, for ```arm-none-eabi-gcc```):

https://pros.cs.purdue.edu/cortex/tutorials/known-issues.html#cannot-find-libstdc-a

Unfortunately, we can't upgrade the students to 16.04 during our school year (production time; can't be done). We're developing 16.04 for our later releases, so hopefully this won't be a problem next year.

#### Actual Behavior:

I downloaded/installed Atom off of GitHub. I didn't use the PROS edition because it depends on ```clang```. I installed ```clang-3.9```, but Atom keeps referencing the ```clang``` package, which in 14.04 is ```clang-3.4```.

```
root@TRUCKEROS-IMG:/opt# sudo dpkg -i pros-editor-amd64.deb 
(Reading database ... 169691 files and directories currently installed.)
Preparing to unpack pros-editor-amd64.deb ...
Unpacking pros-editor (1.21.2) over (1.21.2) ...
dpkg: dependency problems prevent configuration of pros-editor:
 pros-editor depends on clang; however:
  Package clang is not installed.

dpkg: error processing package pros-editor (--install):
 dependency problems - leaving unconfigured
Processing triggers for desktop-file-utils (0.22-1ubuntu1.1) ...
Processing triggers for bamfdaemon (0.5.1+14.04.20140409-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mime-support (3.54ubuntu1.1) ...
Errors were encountered while processing:
 pros-editor
root@TRUCKEROS-IMG:/opt# 
```
I changed the symbolic link for ```clang``` to point to ```3.9```. Here's my ```clang``` information:

```
root@TRUCKEROS-IMG:/opt# /usr/bin/clang --version
clang version 3.9.1-4ubuntu3~14.04.3 (tags/RELEASE_391/rc2)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
root@TRUCKEROS-IMG:/opt# 
```

I have the Cortex connected as described in this diagram:

https://pros.cs.purdue.edu/cortex/tutorials/known-issues.html#linux-a-a-tethering

I see ```ttyACM0``` as an option, and ```lsusb``` reads the device:

```
root@TRUCKEROS-IMG:/opt# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04d8:000a Microchip Technology, Inc. CDC RS-232 Emulation Demo
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@TRUCKEROS-IMG:/opt# 
```

I was wondering if someone could assist. I'll keep working. If I can't get it working, I'll just virtualize Ubuntu 16.04 for our students. Any help & guidance is much appreciated! :)
#### Steps to reproduce:
N/A

#### System information:
Kernel Version: ```4.2.0-42-generic```
```atom``` Version: ```1.32.0```
```git``` Version: ```1.9.1```

#### Additional Information

If I need to move this issue to another repo, please let me know.

norwalk350, 2018-10-29:
Hi @HotelCalifornia :wave:

Yes, I'm trying to open the Cortex serial output via Atom. (BEGIN DIGRESSION) Forgive my lack of knowledge on things VEX, but I'm ultimately trying to use PROS instead of RobotC for our Linux computers. Our 1:1 students use Ubuntu 14.04 as their primary OS. RobotC via Wine doesn't work well. That's how I found out about PROS. 

Could our students use PROS to write & compile their C code without RobotC? I've looked on the VEX forums and it appears PROS can do it. 

https://www.vexforum.com/index.php/28750-pros-vs-robotc/0

IMO, I think our students should use PROS because it reflects real-world applications (e.g. understanding Atom, Linux, protocols, UNIX system calls, etc.) I'm just an IT engineer, I'm not the one actually using this. (END DIGRESSION)

I see ```ttyACM0``` as an option in ```/dev```, and ```lsusb``` reads the device:

```
root@TRUCKEROS-IMG:/opt# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04d8:000a Microchip Technology, Inc. CDC RS-232 Emulation Demo
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
What should I expect when connecting via console to the Cortex? Right now, it just sits and the cursor blinks, like a connection is established. However, no indication of connection is given (e.g. welcome text). If it helps, I can reference one of the students in our robotics team to this thread.

edjubuh, 2018-10-29:
Have you installed the pros-core package as well? That will provide the core binaries PROS uses to create and upload projects you create.

norwalk350, 2018-10-29:
@edjubuh Yes, I downloaded the ```pros-core``` package offered here:

https://github.com/purduesigbots/pros/releases/download/2.12.2/pros-core-2.6.1-amd64.deb

norwalk350, 2018-10-29:
@edjubuh @HotelCalifornia I think I got it. 

Thanks for your assistance! :)

closed 2018-10-29 by norwalk350