purduesigbots/pros #82: Any plan about opensource PROS3 kernel?

FlyGoat, 2019-01-02:
Hi,

Thanks for your hard work on providing a better programming interface for VEX Robotic.
I'm interested in the internals of new V5 controller and wish to make some enhancement on V5, but there are no documents about low-level stuff. The only chance for me is to read PROS open source kernel just as what I did for Cortex. The release notes mentioned that the new kernel will be open-sourced around December but I didn't hear any news about that. 

So I would like to ask if the PROS3 Kernel will be open-sourced? 

Kernel Version: 3.1.0

HotelCalifornia, 2019-01-08:
> So I would like to ask if the PROS3 Kernel will be open-sourced?

yes it will be.

edjubuh, 2019-01-30:
[image]

closed 2019-01-30 by edjubuh