purduesigbots/pros #88: File I/O works with C functions but not with C++ std::fstream

3038922, 2019-01-31:
#### Expected Behavior:
Using STL to read and write files
#### Actual Behavior:
1 if i use "std::fstream" ,file can't open
2 If I use C language, I can open and read files. But if there are Chinese characters, PYTHON3 will make mistakes. Maybe it's the encoding of files.
3 If I set my configRead class as a global class, it will not read correctly, because the global class construction is disorderly, and it only works after initialize (). Because there is no parametric constructor for pros::Motor class, all of my robot project classes can only be constructed as local variables.
#### Steps to reproduce:

#### System information:
Operating System:  (Windows, OS X, Ubuntu)
windows 10 1803
PROS Version: (run `pros --version`)
pros, version 3.1.2
#### Additional Information
kernel 3.3.4
oklib 3.3.7
#### Screenshots/Output Dumps/Stack Traces
[image]
[image]

HotelCalifornia, 2019-03-11:
It seems that the problem is specifically with input streams (`std::ifstream`). Output streams work, but they will only flush when closed (either explicitly or when the dtor is called)

HotelCalifornia, 2019-09-02:
closed by #112

closed 2019-09-02 by HotelCalifornia