purduesigbots/pros #92: Upgrade to LVGL 5.2

HotelCalifornia, 2019-01-31:
#### Expected Behavior:

LVGL is the latest version.

#### Actual Behavior:

LVGL is not the latest version.

#### Steps to reproduce:

Try using 5.2 features (LVGL docs have all been updated to 5.2-specific info) with PROS 😃 

#### System information:
Platform: V5 
PROS Kernel Version: any 

#### Additional Information

This was previously blocked on waiting for a full version of 5.2 to be released, which it [now has been](https://github.com/littlevgl/lvgl/releases/tag/v5.2)

#### Screenshots/Output Dumps/Stack Traces
N/A

3038922, 2019-02-01:
The latest version of LVGL is 5.3

He has increased 
```
lv_spinbox set numeric value with + and - buttons or keys
```
is very nice.

HotelCalifornia, 2019-02-01:
> The latest version of LVGL is 5.3

Ok lol, that happened in the last 12 hours.

3038922, 2019-02-01:
> > The latest version of LVGL is 5.3
> 
> Ok lol, that happened in the last 12 hours.

https://github.com/nlohmann/json
I'm working on a project to automatically generate configuration files based on JSON configuration files.

I serialized std::map

Read the configuration file on the SD card directly after the second boot.

The debugging parameters on the screen are realized without waiting for compilation.

I found arm 7.xx could not be compiled.

I now use GCC version 8.2.1 20181213 (release) [gcc-8-branch revision 267074] (GNU Tools for Arm Embedded Processors 8-2018-q4-major)

It can be compiled

HotelCalifornia, 2019-02-01:
@3038922 that's not relevant to this issue

3038922, 2019-02-04:
> > The latest version of LVGL is 5.3
> 
> Ok lol, that happened in the last 12 hours.

Can you compile separate LVGL static libraries? If LVGL is updated, it's convenient for us to update.

HotelCalifornia, 2019-02-04:
> Can you compile separate LVGL static libraries? If LVGL is updated, it's convenient for us to update.

Good question. We have considered doing this, but have decided against it (at least in the short-term). It may be revisited at some point in the future.

PR #144 (2019-08-19): ⬆️ Upgrade LVGL to 5.3
#### Summary:
Upgrades LVGL to 5.3 release

#### Motivation:
everyone reeeealllly wants this

##### References (optional):
Should unblock #110 and by extension #93
Closes #92 

#### Test Plan:
- [x] let people play with it i guess
closed 2019-08-19 by HotelCalifornia