purduesigbots/pros #101: Allow copying text from he terminal window

graham615, 2019-02-07:
#### Expected Behavior:
Using printf statements to output telemetry to the terminal window I had hoped the data could be copied to the clipboard using cut/copy/paste.

#### Actual Behavior:
When the text was selected in the window and copied, when pasted in another window, no test appeared.

#### Steps to reproduce:
printf output to the terminal
select the text in the terminal pane
select Edit->copy
in a unrelated application ( notepad), Excel, select paste
no text is pasted

#### System information:
Platform: V5
PROS Kernel Version:1.34

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2019-02-07:
What terminal window are you referring to? Is this in the PROS Editor?

graham615, 2019-02-07:
Alex:

Yes.  Any work around?  Command line perhaps?  Miss robots debugger but access to a real compiler is worth it. 

Graham

HotelCalifornia, 2019-02-07:
yes, if you open the command palette (ctrl+shift+P on windows/linux and cmd+shift+P on macOS), you can search for the command `terminal:copy`. Running that command will copy whatever text is currently selected in the terminal tab.

It looks like the package we use provides a default keybinding for this command on macOS (cmd+C), but not for windows/linux. You could add a binding to your keymaps.cson (Settings > Keybindings > click the "your keymap file" link) like this:

```cson
'terminal-view':
  'ctrl-alt-c': 'terminal:copy'
  'ctrl-alt-v': 'terminal:paste
```