purduesigbots/pros #103: Why is compilation so slow?

3038922, 2019-02-10:
I just tried to compile on Ubuntu 18.04, and the compilation speed is still very slow.
Is it makefile?

edjubuh, 2019-02-10:
No, it's these lines:

https://github.com/purduesigbots/pros/blob/f34a219cb49a6b5a8e81b6852abe4c84efa975d0/common.mk#L69-L75

I've thought about how to avoid writing to temp.log and temp.errors, but haven't been able to get around to actually writing/testing anything. If someone wants to submit a tested PR for it, that'd be much appreciated.

3038922, 2019-02-11:
I found the Ninja tool in the process of installing ccls, which compiled llvm for windows much faster than make.
My idea is to use Ninja instead of make.
Ninja's writing grammar is abstract, so I use cmake to generate rules first.
At present, there are still some problems with my cmakeLists. If I succeed, I will share them.

PR #123 (2019-03-11): ✨ Improve Makefile compile speed
#### Summary:
Someone doesn't want to waste his time. So he adds this code to make compile-job faster. The compiler now doesn't need to remove and create `temp.log` every time it compiles a file. 

#### Motivation:
Add it.

##### References (optional):
close #103 

#### Test Plan:
- [x] Call `make all`
- [x] Call `make`
- [x] Try to compile with no error
- [x] Try to compile with warning
- [x] Try to compile with error
commit 021f8bec (2019-03-11): ✨ Improve Makefile compile speed (#123)

#### Summary:
Someone doesn't want to waste his time. So he adds this code to make compile-job faster. The compiler now doesn't need to remove and create `temp.log` every time it compiles a file. 

#### Motivation:
Add it.

##### References (optional):
close #103 

#### Test Plan:
- [x] Call `make all`
- [x] Call `make`
- [x] Try to compile with no error
- [x] Try to compile with warning
- [x] Try to compile with error

#### Commits:

* ✨ Improve Makefile compile speed

* Use only variables; use printf

* Keep changes from #119

* Do not treat strings a [... cut]
closed 2019-03-11 by edjubuh