purduesigbots/pros #104: Add steps for compiling the kernel

Skyluker4, 2019-02-11:
It is possible for anyone to compile the kernel, it just requires a few extra steps. VCS actually ships with the SDK and can be added to PROS.

### Steps:
1. Download Vex Coding Studio
2. Copy the libv5rt header files (the ones prefixed with "**v5_**") from where the VCS SDK is installed to **[pros source folder]/firmware/libv5rts/sdk/vexv5/include/**

- In Windows they are in **%localappdata%/VEX Coding Studio/VEX Coding Studio/sdk/vexv5/include/**

3. Copy the prosv5rt library (**libv5rt.a**) to **[pros source folder]/firmware/libv5rts/sdk/vexv5/** and rename it to **libv5rts.a**
- In Windows it is located in the folder **%localappdata%/VEX Coding Studio/VEX Coding Studio/sdk/vexv5/**

4. Define the function vexBackgroundProcessing defined in **v5_api.h**.
- Create the source file **[pros source folder]/src/libv5rts/sdk/vexv5/v5_api.c** and add the following code:
```
#include "v5_api.h"

#pragma GCC push_options
#pragma GCC optimize("O1")
void vexBackgroundProcessing(void) {
    __asm__ ("ldr r3, [pc, #4]");
    __asm__ ("ldr r3, [r3, #92]");
    __asm__ ("bx r3");
    __asm__ ("cmneq pc, #0");
}
#pragma GCC pop_options
```

- and edit **v5_api.h** so that the function declaration is 
```
void                  vexBackgroundProcessing( void ) __attribute__((naked));
```

5. Compile with _prosv5 make_

### Solution
Update the readme to include steps on how to compile the kernel and possibly add the code above to the repo so anyone can build the kernel themselves.

Jerrylum, 2019-02-11:
[image]
It is work, at least in the compiler
I haven't used in the Brain yeah.

Jerrylum, 2019-02-11:
### Steps:
1. Download Vex Coding Studio and run the program at least once
> so the Studio can init itself.

2. Copy the `libv5rt header files`  to `[pros source folder]\include\`
> libv5rt header files is in **%appdata%\Local\VEX Coding Studio\VEX Coding Studio\sdk\vexv5\include**. You only need files with "v5_" prefixed. (`v5.h` to `v5_vcs_options.h`)

3. Copy the `libv5rt.a` to `[pros source folder]\firmware\libv5rts\sdk\vexv5`, and rename it to `libv5rts.a`
> libv5rt.a is in **%appdata%\Local\VEX Coding Studio\VEX Coding Studio\sdk\vexv5**

4. Create file `v5_api.c` in  `[pros source folder]\src\system\` and add the following code:
```cpp
#include "v5_api.h"

#pragma GCC push_options
#pragma GCC optimize("O1")
void vexBackgroundProcessing(void) {
    __asm__ ("ldr r3, [pc, #4]");
    __asm__ ("ldr r3, [r3, #92]");
    __asm__ ("bx r3");
    __asm__ ("cmneq pc, #0");
}
#pragma GCC pop_options
```

5.  Edit `v5_api.h` so that the function declaration is
```cpp
void                  vexBackgroundProcessing( void ) __attribute__((naked));
```
> v5_api.h is in the **[pros source folder]\include\**

6. Compile with `prosv5 make`

Jerrylum, 2019-02-11:
[image]
[image]

I tested about it, you can use the raw method on `VEX CODE STUDIO` but `PROS` can't. Also the library have `VEX CODE STUDIO`'s methods
*raw method mean The super security **VEX's proprietary Software Development Kit**

### What it mean?
you can write `VCS` code on the `Notepad++` and comiple or upload it using `PROS CLI`. OR
write your own code without `PROS kernel`. OR
add `VCS` support on the `PROS Kernel`. OR
use `PROS API` and `SDK` in your project.

Finally, it is no reason to private the `libv5rt.a` right now.

HotelCalifornia, 2019-02-11:
We do not distribute the SDK because it is proprietary and owned by VEX. We can't stop you from doing this on your own, but we're also not going to support it.

closed 2019-02-11 by HotelCalifornia