purduesigbots/pros #105: Gyro Instability

graham615, 2019-02-12:
#### Expected Behavior:
We expect the gyro to return a reasonably stable value after initialization with the gyro held at fixed orientation.

#### Actual Behavior:
After initialization and with the gyro held still we see increasing values returned from the gyro.  As we thought this was a hardware issue, we ordered a replacement and experienced the same result.  We have changed ports with no difference in behavior.

#### Steps to reproduce:

**At Drive.cpp file scope:**

#define GYRO_SENSOR        ('A')

// a pointer to the gyro - we will need to create it later
pros::ADIGyro *dtGyro;

**Drive.cpp drivetrainInit(), called from initialize:**

   // create our gyro - it will get initialized here so we need to be in position
   dtGyro = new pros::ADIGyro( GYRO_SENSOR );

   // wait a second
   pros::delay(1000);

**Drive.cpp arcadeMove - called every 10ms:**

   //gyro test
   if (timer3++  > 100)
   {
     printf("Gyro = %f\n",  drvGetDirection( ));
      timer3 = 0;
   }

float drvGetDirection( void )
{
   return dtGyro->get_value();
}

#### System information:
Platform: V5
PROS Kernel Version: 1.3.4

#### Additional Information
The first gyro showed increasing values or second, totally new gyro ( right out of the box) increased in increments of 100.  I 

#### Screenshots/Output Dumps/Stack Traces

baylessj, 2019-02-12:
With V5, our access to the gyroscope is significantly limited compared to PROS for Cortex. On V5, the integration for the gyroscope is done on the ADI itself, not in PROS kernel. We theoretically could write a gyroscope integrator again in the same manner as PROS for Cortex, but since we only get new values from the ADI every 10ms the performance would likely be worse than the default behavior.

The behavior you are experiencing is almost definitely the result of improper calibration, which is done at the VexOS level. Make sure that the gyroscope is completely still during the delay after constructing the ADIGyro object. The gyroscope driver on V5 is very sensitive to gyro movement during calibration, I experienced the same behavior that you are reporting during testing if I wasn't very careful to keep the gyro stable.

closed 2019-02-12 by baylessj

graham615, 2019-02-12:
Thank you Jonathan.  One last question on this topic.  Do you know if VCS has this limitation?  Are both PROS and VCS build on the same framework?  We need to make a determination if a move to VCS would solve this issue.  Under two weeks to State....just qualified Sunday after an, up till now, dismal season.

 
 
Closed #105.

baylessj, 2019-02-12:
Both PROS and VCS use the same gyroscope driver, you'll get very similar results for the gyro with both platforms. My recommendation would be to calculate turning distance using the integrated encoders on your drive motors instead of using the gyroscope. This is the method that [OkapiLib](https://pros.cs.purdue.edu/v5/okapi/) uses for its ChassisController turns, which works quite well for a simple approach.

graham615, 2019-02-12:
Thank you.  Appreciate your input....saving us a ton of experimentation.  We do have solid turning code but hoped to improve our repeatability using a sensor.  Perhaps VEX will have an updated gyro for next year.  All the best and keep up the good work as we will be staying with pros.