purduesigbots/pros #108: Project Upgrade: cmd.exe could not be spawned

graham615, 2019-02-14:
#### Expected Behavior:
Pros and okapi upgraded to the latest versions

#### Actual Behavior:
Red Error Box containing:
'C:\WINDOWS\system32\cmd.exe' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.

#### Steps to reproduce:
PROS->Upgrade Project
Point to top level project directory
Select upgrade to latest
Verified path and existence of cmd.exe.  Opened pros in Admin.

#### System information:
Platform: 
PROS Kernel Version: V% 3.1.5

#### Additional Information
kage spawning the process.
[image]

#### Screenshots/Output Dumps/Stack Traces

Jerrylum, 2019-02-14:
It is about the Atom toolbar, not pros-atom and any other component. right? @HotelCalifornia

graham615, 2019-02-14:
New in town and don't know the difference but.....

HotelCalifornia, 2019-02-14:
> Atom toolbar, not pros-atom and any other component. right? @HotelCalifornia

I'm not really sure what you mean. This looks like an issue with the environment, but we're still trying to figure out what's causing it