purduesigbots/pros #133: Line Sensor Error

lebraddock, 2019-03-17:
Expected Behavior:
The line sensor should print a value to the screen between 0 and 4095

Actual Behavior:
The printed value is -3453454........  The value is so large that it goes of the screen. If I use get_value_calibrated() then 0 is printed. This occurs on both line sensors and multiple ports 

Platform: v5 using the latest kernel

[image]

baylessj, 2019-03-17:
Can you post your test code? This looks indicative of using `%d` when printing a float or vice versa. I've recently confirmed that line sensors work on the latest version of the kernel so it is unlikely that the issue is in the kernel.

lebraddock, 2019-03-17:
`pros::lcd::print(0, "Value: %f", fireSenseLow.get_value());`

baylessj, 2019-03-17:
Try using %d instead, a line sensor returns an integer (https://pros.cs.purdue.edu/v5/api/cpp/adi.html#get-value)

closed 2019-03-17 by baylessj

lebraddock, 2019-03-17:
that was it thank you