purduesigbots/pros #134: GCC 8 error with __sync_synchronize

lights0123, 2019-03-17:
#### Expected Behavior:
Successful compilation

#### Actual Behavior:
An error message referencing `__sync_synchronize` appears (see below).

#### Steps to reproduce:
I had created a new project with `prosv5 conductor new-project test`, after installing `arm-none-eabi-gcc 8.3.0`.

#### System information:
Platform: V5
PROS Kernel Version:

|  Name  | Version |       Origin       |
|-----------|-----------|-------------------|
|  kernel  |   3.1.6   | pros-mainline |
| okapilib |  3.3.12  | pros-mainline |

#### Additional Information
I am running Arch Linux, using `arm-none-eabi-gcc (Arch Repository) 8.3.0`. Is this a GCC 8 issue? If so, that should be fixed or added to the documentation.

#### Screenshots/Output Dumps/Stack Traces
```c++
❱ make
Adding timestamp [OK]
Linking project with libpros,okapilib [ERRORS]
/usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/lib/libstdc++.a(cxx11-ios_failure.o): in function `(anonymous namespace)::__io_category_instance()':
cxx11-ios_failure.cc:(.text._ZN12_GLOBAL__N_122__io_category_instanceEv+0xc): undefined reference to `__sync_synchronize'
/usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/lib/libstdc++.a(locale_init.o): in function `(anonymous namespace)::get_locale_mutex()':
locale_init.cc:(.text._ZN12_GLOBAL__N_116get_locale_mutexEv+0xc): undefined reference to `__sync_synchronize'
/usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/lib/libstdc++.a(locale.o): in function `std::locale::_Impl::_M_install_cache(std::locale::facet const*, unsigned int)':
locale.cc:(.text._ZNSt6locale5_Impl16_M_install_cacheEPKNS_5facetEj+0x18): undefined reference to `__sync_synchronize'
collect2: error: ld returned 1 exit status
make: *** [common.mk:191: bin/monolith.elf] Error 1
Error: Failed to build
```

edjubuh, 2019-03-18:
I know we support the latest [official release from Arm](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads), so that'd be 8.2.1. I've been using that toolchain version for the past couple weeks on Windows. I'm not sure where you're getting a toolchain with version 8.3.0 from. It's unlikely that we can support that toolchain as you've observed.

lights0123, 2019-03-18:
It is [built from GCC's sources](https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/arm-none-eabi-gcc), not an official GCC build. I'll probably hack together something with Docker (or adjust my PATH based on the directory) to use the official build.

HotelCalifornia, 2019-03-28:
feel free to reopen this issue if you have more problems with GCC 8 (using the official build)

closed 2019-03-28 by HotelCalifornia

sealj553, 2019-05-01:
I was experiencing this issue as well on Arch, same gcc version.

Solution:
In `common.mk:4`: change `-mcpu=cortex-a9` to `-march=armv7-a`