purduesigbots/pros #136: Failed to load the terminal-tab-plus package。。。。。。。another pc

vex-lq, 2019-03-28:
[image]

vex-lq, 2019-03-28:
reinstalling does not help at all

vex-lq, 2019-03-28:
does std::cout work with PROS??

HotelCalifornia, 2019-03-28:
> does std::cout work with PROS??

yes

vex-lq, 2019-03-28:
how? i  dont see ahy outputs in terminal by std::cout

vex-lq, 2019-03-28:
terminal-tab-plus now is working, i copied this package from another pc, but std::cout is not working..and is there a debugger?

Technical27, 2019-04-27:
The output of `std::cout` can be seen in PROS terminal, which you can use by plugging in your V5 to your computer and running `pros terminal` or `prosv5 terminal`

HotelCalifornia, 2019-04-29:
Alternatively, you can click this button in the PROS editor:

[image]

or use the menus:

[image]

If you are still having problems, please feel free to reopen this issue.

closed 2019-04-29 by HotelCalifornia