purduesigbots/pros #137: how do i print simple info like Brain.Screen.print  or printAt in vexCode? thanks

vex-lq, 2019-03-30:
how do i print simple info like Brain.Screen.print  or printAt in vexCode? thanks

vex-lq, 2019-03-30:
i dont want to use the emulated LCD screen

Technical27, 2019-04-27:
PROS uses LVGL to control the screen on the V5 brain. You first need to include the `lvgl.h` header in your `main.h` header file.
Open the file `include/main.h` in a text editor. scroll down until you see a section like this:
```
/**
 * You should add more #includes here
 */
```
Below that section, paste the following line: `#include "display/lvgl.h"`
____
Now, you can use LVGL in your code.
Here is an commented example where it puts some text at the center of the screen:
```
// Creates the label
lv_obj_t* text = lv_label_create(lv_scr_act(), NULL);

// Sets the text of the label
lv_label_set_text(text, "this is the text of the label");

// Change width of the label
lv_obj_set_width(text, 300);

// Align to the center of the screen
lv_obj_align(text, NULL, LV_ALIGN_CENTER, 0, 0);
```
Put this code in the `initialize` function and you should have something on the screen. You can look at the LVGL documentation to see all of the GUI elements you can create. You can click [here](https://docs.littlevgl.com) for the documentation. I hope this helps you.

HotelCalifornia, 2019-04-29:
> Below that section, paste the following line: `#include "display/lvgl.h"`

Actually, you can just `#include "pros/apix.h"`

HotelCalifornia, 2019-04-29:
I'm going to close this, as it is not actually a bug report

closed 2019-04-29 by HotelCalifornia