purduesigbots/pros #140: Fix header issues with user-defined headers

Technical27, 2019-04-27:
User-defined headers don't link properly, ignore header guards and `#pragma once`, and source files don't recompile when a header changes. Header guards and `#pragma once` are both broken, because all source code files are complied separately, and not together.

Also when a header changes, source code that `#include`s the header don't get recompiled.

The solution is to compile all source code files at the same time, and then header guards and `#pragma once` will work.
The next solution is that if a header changes, then to recompile all source code that `#include`s it.

There is a workaround with the second problem, and that is to change the `DEFAULT_GOAL` in your `Makefile` to `all` instead of `quick`. This recompiles all files every time and fixes the issue.

This issue has been discussed in issue #81, but I wanted to include the fact that header guards don't work.

edjubuh, 2019-04-27:
Could you provide a more concrete example of header guards not working? We use [`#pragma once`](https://github.com/purduesigbots/pros/blob/develop/include/common/cobs.h#L16) and [`#ifndef`](https://github.com/purduesigbots/pros/blob/develop/include/api.h#L19) style header guards in the kernel and both compile just fine.

Getting header file dependencies included in the Makefile is on our backlog but isn't something I realistically see us working on in the near future due to other higher priority tasks. This is definitely something that I think a community member could contribute to though, and we'd welcome a pull request. @PIxelToast had a good start to it in #81 but we ended up making significant changes to the Makefile before the PR was merged and it just never got updated.

Technical27, 2019-04-27:
You can reproduce this by creating your own header file and `#include` it. In the header put something like:
```
#include "main.h"
extern int foo;
#pragma once
int foo = 1;
```
Here is the output of `make all`:
```
Cleaning project
Adding timestamp [OK]
Creating cold package with libpros,okapilib [OK]
Stripping cold package  [DONE]
Section sizes:
   text	   data	    bss	  total	    hex	filename
999.94KB  3.43KB  47.59MB  48.57MB 3091305 bin/cold.package.elf
Compiling src/autonomous.cpp [OK]
Compiling src/initialize.cpp [OK]
Compiling src/opcontrol.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and libpros,okapilib [ERRORS]
/opt/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: bin/initialize.cpp.o:(.data.foo+0x0): multiple definition of `foo'; bin/autonomous.cpp.o:(.data.foo+0x0): first defined here
/opt/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: bin/opcontrol.cpp.o:(.data.foo+0x0): multiple definition of `foo'; bin/autonomous.cpp.o:(.data.foo+0x0): first defined here
collect2: error: ld returned 1 exit status
common.mk:213: recipe for target 'bin/hot.package.elf' failed
make: *** [bin/hot.package.elf] Error 1
```
All files compile, but `ld` complains that `foo` is defined multiple times. If you define `foo` in any source code file such as `initialize` and then remove the `#pragma once` and the definition of `foo` in the header file, then it compiles normally.
Now, the output of `make all` is:
```
Cleaning project
Adding timestamp [OK]
Creating cold package with libpros,okapilib [OK]
Stripping cold package  [DONE]
Section sizes:
   text	   data	    bss	  total	    hex	filename
999.94KB  3.43KB  47.59MB  48.57MB 3091305 bin/cold.package.elf
Compiling src/autonomous.cpp [OK]
Compiling src/initialize.cpp [OK]
Compiling src/opcontrol.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and libpros,okapilib [OK]
Section sizes:
   text	   data	    bss	  total	    hex	filename
676.00B  12.00B   3.00B 691.00B     2b3 bin/hot.package.elf
Creating cold package binary for VEX EDR V5 [DONE]
Creating bin/hot.package.bin for VEX EDR V5 [DONE]
```
You can also test a header guard just like this:
```
#include "main.h"
extern int foo;

#ifndef _INCLUDE_HPP_
#define _INCLUDE_HPP_
int foo = 1;
#endif
```
and `make all` will show the same output as `#pragma once`

Technical27, 2019-04-27:
I forgot to mention that you need to `#include` the header in 2 or more files. I also have reproduced this issue in a completely new project following my instructions. It also has the same behavior I described earlier.

edjubuh, 2019-04-27:
That's not the correct usage of headers. See this [forum thread](https://www.vexforum.com/t/pros-code-not-working/60355).

Technical27, 2019-04-27:
Ah, ok then I'll just close this issue

closed 2019-04-27 by Technical27