purduesigbots/pros #145: When sitting idle my brain crashes and my motor lights flicker

lerker100, 2019-05-23:
#### Actual Behavior:
On occasion when the robot is on sitting idle running code the entire robot crashes.
The only way to reset it is to unplug the battery.

#### Steps to reproduce:
Not quite sure but the code that I'm running is here.
It's A WIP so don't judge it to hard please.
I was having this same problem last year but wasn't convinced that it wasn't my code until I started a new project for the new season.
https://github.com/lerker100/Nebula-2020-PROS

#### System information:
Platform:V5
PROS Kernel Version:3.1.6

HotelCalifornia, 2019-05-23:
Right off the bat, I see that you don't have a delay in your opcontrol loop. I'm not convinced that's causing the specific issues you've observed, but starving the processor in the way (i.e. denying other tasks processing time) will cause undesirable behavior.

HotelCalifornia, 2019-05-23:
Ditto for the telemetry task, which appears to be an empty loop (again, no delay)

lerker100, 2019-05-23:
I uploaded the wrong folder give me a sec.

lerker100, 2019-05-23:
It's updated now.

HotelCalifornia, 2019-05-23:
Okay, nothing seems glaringly wrong as far as your code goes.

Does this phenomenon happen repeatedly? If so, is it at some fixed interval?

lerker100, 2019-05-23:
Yes, but it's kinda random, Some times it will be fine for weeks and other times it will do it multiple times a week.

HotelCalifornia, 2019-05-23:
I'm _pretty_ sure there's no inherent problem with PROS that's causing this.

My best guess is that it's something to do with low battery charge maybe?

Since there's no way to reliably reproduce this (and to my knowledge there haven't been any other reports of this behavior), I'm going to close the issue. If you do figure out a way to reproduce, please feel free to reopen.

In the meantime, I'd suggest posting something about it on the vex forum, on the off-chance that someone else has actually run into this.

closed 2019-05-23 by HotelCalifornia