purduesigbots/pros #146: Can't compile on Arch Linux

IrregularWombat, 2019-06-16:
#### Expected Behavior:

For the code to compile with out any errors and give me a binary which can be uploaded to a V5 Brain.

#### Actual Behavior:

Fails to compile giving me the error `Linking project with libpros,okapilib [ERRORS]`

#### Steps to reproduce:

Install latest version of Arch Linux, install relevant ARM and GCC packages, use `prosv5 conductor new-project ./test` code fails to compile.

#### System information:
Platform:  V5
PROS Kernel Version: 
```
PROS Project for v5 at: /home/milo/help (help)
Name      Version    Origin
--------  ---------  -------------
kernel    3.1.6      pros-mainline
okapilib  3.3.13     pros-mainline
```

#### Additional Information

This was happening on my system and after a while of messing around I was not able to fix it, I reinstall a fresh copy Arch Linux and still get this problem. I am able to compile other pieces of code for ARM on this machine as well.
My mate was able to compile just fine on a Ubuntu system, perhaps it may be that Arch has newer version of ARM GCC compilers which cause this error?

#### Screenshots/Output Dumps/Stack Traces
Level 0 Verbosity from `prosv5 make`
```
Adding timestamp [OK]
Linking project with libpros,okapilib [ERRORS]
/usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/lib/libstdc++.a(locale.o): in function `std::locale::_Impl::_M_install_cache(std::locale::facet const*, unsigned int)':
locale.cc:(.text._ZNSt6locale5_Impl16_M_install_cacheEPKNS_5facetEj+0x18): undefined reference to `__sync_synchronize'
/usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/lib/libstdc++.a(locale_init.o): in function `(anonymous namespace)::get_locale_mutex()':
locale_init.cc:(.text._ZN12_GLOBAL__N_116get_locale_mutexEv+0xc): undefined reference to `__sync_synchronize'
/usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/lib/libstdc++.a(cxx11-ios_failure.o): in function `(anonymous namespace)::__io_category_instance()':
cxx11-ios_failure.cc:(.text._ZN12_GLOBAL__N_122__io_category_instanceEv+0xc): undefined reference to `__sync_synchronize'
collect2: error: ld returned 1 exit status
make: *** [common.mk:191: bin/monolith.elf] Error 1
```
Level 2 Verbosity from `prosv5 make`
```
Adding timestamp echo 'char const * const _PROS_COMPILE_TIMESTAMP = __DATE__ " " __TIME__; char const * const _PROS_COMPILE_DIRECTORY = "/home/milo/help";' | arm-none-eabi-gcc -c -x c -mcpu=cortex-a9 -mfpu=neon-fp16 -mfloat-abi=softfp -D_POSIX_THREADS -D_UNIX98_THREAD_MUTEX_ATTRIBUTES -Os   -ffunction-sections -fdata-sections -fdiagnostics-color --std=gnu11  -o ./bin/_pros_ld_timestamp.o - 2> temp.log || touch temp.errors
[OK]
Linking project with libpros,okapilib arm-none-eabi-g++ -mcpu=cortex-a9 -mfpu=neon-fp16 -mfloat-abi=softfp   -nostdlib ./bin/autonomous.cpp.o ./bin/initialize.cpp.o ./bin/opcontrol.cpp.o ./bin/_pros_ld_timestamp.o -Wl,-T./firmware/v5.ld,--gc-sections,--start-group,./firmware/libpros.a,./firmware/okapilib.a,-lc,-lm,-lgcc,-lstdc++,-lsupc++,--end-group -o bin/monolith.elf 2> temp.log || touch temp.errors
[ERRORS]
/usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/lib/libstdc++.a(locale.o): in function `std::locale::_Impl::_M_install_cache(std::locale::facet const*, unsigned int)':
locale.cc:(.text._ZNSt6locale5_Impl16_M_install_cacheEPKNS_5facetEj+0x18): undefined reference to `__sync_synchronize'
/usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/lib/libstdc++.a(locale_init.o): in function `(anonymous namespace)::get_locale_mutex()':
locale_init.cc:(.text._ZN12_GLOBAL__N_116get_locale_mutexEv+0xc): undefined reference to `__sync_synchronize'
/usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.1.0/../../../../arm-none-eabi/lib/libstdc++.a(cxx11-ios_failure.o): in function `(anonymous namespace)::__io_category_instance()':
cxx11-ios_failure.cc:(.text._ZN12_GLOBAL__N_122__io_category_instanceEv+0xc): undefined reference to `__sync_synchronize'
collect2: error: ld returned 1 exit status
make: *** [common.mk:191: bin/monolith.elf] Error 1
Error: Failed to build
```
List of installed packages on system.
[install packages.txt](https://github.com/purduesigbots/pros/files/3293781/install.packages.txt)

HotelCalifornia, 2019-06-16:
Yeah, I'm pretty sure we've answered this before. The version of arm-none-eabi-gcc on the arch package repository is some weird custom-built version. We only guarantee support for the official builds which can be found [here on the Arm website](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads).

closed 2019-06-16 by HotelCalifornia

IrregularWombat, 2019-06-16:
Ah sorry I didn't see it, thanks I'll go install that version.