purduesigbots/pros #150: C++ exceptions not working with hot/cold linking

vexcat, 2019-07-14:
#### Expected Behavior:
To be able to throw and catch an exception without `terminate()` being called, regardless of whether hot/cold linking is enabled.

#### Actual Behavior:
On hot/cold linking, throwing a `std::runtime_error` and catching it with `catch(...)` will result in this output:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Exception text
```

#### Steps to reproduce:
1. Create a new PROS project
2. In `initialize()` write
```
void initialize() {
	try {
		pros::delay(5000);
		throw std::runtime_error("Exception text");
	} catch(...) {
		printf("Caught some error.\n");
	}
}
```
3. Build and run to get expected output
4. Change USE_PACKAGE to 1 in Makefile
5. Build and run to get unexpected output
#### System information:
Platform: V5
PROS Kernel Version: 3.1.6

#### Additional Information
Using `arm-none-eabi` 7.3.1.

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2019-07-14:
I bet this is related to what I'm seeing happen in #112

edjubuh, 2019-09-01:
Tested this with the changes in #112 in hot/cold and I see `Caught some error.` being printed.

PR #112 (2019-09-02): 🚨🔊 Add a better Data Abort Handler
#### Summary:
Dunno how to be concise yet. But this makes the data abort handler do something interesting.
- We have our own error box on the V5 that reports the proper PC and also current task name (because we can)
- We still background process and printfs work, so I print registers of the state that caused the abort and stack trace of the running task
   - This required adding unwind tables. Some additional adjustments will need to be made once #89 lands to ensure that the unwinder can find both hot and cold tables.

There will need to be additional work in [... cut]
closed 2019-09-02 by HotelCalifornia