purduesigbots/pros #153: Linking fails with Arm Toolchain 8-2019-q3-update Linux 64-bit

aderic6, 2019-07-28:
#### Expected Behavior:
Ran prosv5 make all and expected to compile into hot and cold image.

#### Actual Behavior:
Compiling of source files works until attempting to link with libpros and libokapi. The same set of files compile with the previous 2018 December Arm Toolchain.

#### Steps to reproduce:
1. Download the new Linux 64-bit toolchain from [Arm Developers](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads)
2. Extract and move the folder to /opt/arm-gcc-none-eabi/ replacing old toolchain
3. Export to PATH using ~/.profile and sourcing the file.
4. Make a new PROS project with conductor.
5. Run prosv5 make in project directory (both with and without hot and cold linking yields the same error).

#### System information:
Manjaro 18.0.4 (4.19.60-1-MANJARO)

PROS Version: pros, version 3.1.4

#### Additional Information
Ran on a clean install with no other conflicting packages and whereis points to the extracted toolchain. The linking failure occurs with the default PROS project source files too.
Python 3.7.3 used and PROS installed using pip3 without --user flag.

#### Screenshots/Output Dumps/Stack Traces
[image]

HotelCalifornia, 2019-07-28:
Done some looking, and it seems like [you can just stub it ](https://forums.xilinx.com/t5/Embedded-Development-Tools/mfloat-abi-soft-causes-link-error-in-C-projects/td-p/909166) in the interim. It also seems like we might be able to solve this more effectively by playing with the arch flags-- I don't like just adding a stub as a solution unless I can say for sure that we don't need the symbol to do anything, especially if we want to get pthread working.

HotelCalifornia, 2019-08-08:
ok, done some more looking and found [this thread](https://e2e.ti.com/support/tools/ccs/f/81/t/713517) which suggests defining the stub like so:

```c
void __sync_synchronize(void) {
    __sync_synchronize();
}
```

because

> ... some of the libstdc++ librararies in [arm-none-eabi] are broken because they try and call a function named __sync_synchronize(), instead of have being compiled using the built-it of that name.

and 

> this looks like an infinite recursion, but looking at the generated assembler shows that the compiler did generate a function named __sync_synchronize which contains the memory barrier instruction

I added this to the newlib_stubs.c file, which fixed the linking error. Checking the cold package with arm-none-eabi-objdump reveals that

```
0383717c <__sync_synchronize>:                                                                                                                                                                                                                 
  383717c:       f57ff05b        dmb     ish                                                                                                                                                                                                    
  3837180:       e12fff1e        bx      lr
```

where `dmb ish` is the correct memory barrier instruction. Will open a PR shortly.

PR #155 (2019-08-18): 🚨 Make things work with GCC 8.3
#### Summary:

arm-none-eabi-8-2019-q3-update was released, which finally bumped arm-none-eabi-gcc to 8.3 officially, so we started running into the missing `__sync_synchronize` symbol as well. This PR fixes that, as well as a new multiple definitions issue (`__malloc_lock`/`unlock`).

The solution to the `__sync_synchronize` issue is described in https://github.com/purduesigbots/pros/issues/153#issuecomment-519335375.

`__malloc_lock` and `__malloc_unlock` were moved from src/system/newlib_stubs.c to their own file src/system/mlock.c, which [apparently is how n [... cut]
closed 2019-08-18 by HotelCalifornia