purduesigbots/pros #159: clam_port macro used in functions that returns a double

Jerrylum, 2019-08-29:
According to lines 47 ~ 48 in /include/api.h
https://github.com/purduesigbots/pros/blob/develop/include/api.h#L47-L48

As far as I know, the first one (`PROS_ERR_F`) is used in functions which return an integer value and `PROS_ERR_F`is used in functions which return a double value.

But according to the file /src/devices/vdml_motors.c and file /include/vdml/vdml.h 
https://github.com/purduesigbots/pros/blob/develop/src/devices/vdml_motors.c#L178-L182

https://github.com/purduesigbots/pros/blob/develop/include/vdml/vdml.h#L28-L57

The `claim_port` macro returns `PROS_ERR`. It used the claim_port in all functions but not all of them should return `PROS_ERR`.

#### Expected Behavior:
claim_port macro should only be used in functions returning int32_t, so we need to change `claim_port` behavior.

#### Steps to reproduce:
Try it yourself

#### System information:
Platform: V5
PROS Kernel Version: all

#### Screenshots/Output Dumps/Stack Traces

baylessj, 2019-08-29:
Good catch, thanks for opening this issue. Looks like we used just the `internal_port_mutex_take` function by itself to return PROS_ERR_F here: https://github.com/purduesigbots/pros/blob/65a5845a95fb8a79db5147646d754a53d42be6a4/src/devices/battery.c#L38

The simplest fix would be to use this exact same method for the functions that you mentioned that are still using `claim_port` but it might be worth creating a `claim_port_f` macro for these instances.

Octogonapus, 2019-09-19:
@HotelCalifornia This should be closed.

HotelCalifornia, 2019-09-19:
Closed in #160. Not sure why it didn't close this automatically

closed 2019-09-19 by HotelCalifornia