purduesigbots/pros #161: Motor's PID function problem

Jerrylum, 2019-08-30:
According to lines 26 ~ 46 In the `/vdml/vdml_motors.c`

[image]

As you can see, 4 PID functions here. They are using a function call `registry_get_device`

[image]
And this is the user-end function.

But I found that we should put a 0-based number to use this function. In this case, the function is not controlling the right Smart Port.

#### Expected Behavior:
The example of using `registry_get_device` function:
[image]

[image]
The `claim_port` function, at `/include/vdml/vdml.h` .

#### Steps to reproduce:
Try it yourself

#### System information:
Platform: V5
PROS Kernel Version: all

Octogonapus, 2019-09-19:
@HotelCalifornia This should be closed.

HotelCalifornia, 2019-09-19:
Closed in #162. Not sure why it wasn't closed automatically

PR #162 (2019-09-15): 🐛 Fix motor PID functions bug
#### Summary:
Someone forgot to check the port number. It should 0-based numbering.

#### Motivation:
Love? Maybe...

##### References (optional):
closes #161;  #160 

#### Test Plan:
- [x] Run test code from https://github.com/purduesigbots/pros/pull/162#issuecomment-531485060
closed 2019-09-19 by HotelCalifornia