purduesigbots/pros #171: terminal ouput from opcontrol appearing before banner

theol0403, 2019-09-19:
This issue is very minor and low priority.
#### Expected Behavior:
The PROS banner should show in the terminal, then any user text should be printed after the banner.

#### Actual Behavior:
User text is printed before or in the middle of the banner:

[image]

#### Steps to reproduce:
```cpp
void opcontrol() {
	std::cout << "This print is at the top of opcontrol()" << std::endl;
	while (true) {
		pros::delay(20);
	}
}
```
#### Solution
One solution would be to delay about 200ms before starting the competition control tasks.
This will have the added benefit of giving the ADI ports time to initialize.

#### System information:
PROS Kernel Version: 3.2.0

#### Additional Information
This is strange behavior, considering
https://github.com/purduesigbots/pros/blob/6394b233ac02e337cd4d8dc103a8c7f069272ae4/src/system/startup.c#L36-L52
and
https://github.com/purduesigbots/pros/blob/6394b233ac02e337cd4d8dc103a8c7f069272ae4/src/system/system_daemon.c#L125

Octogonapus, 2019-09-19:
> This will have the added benefit of giving the ADI ports time to initialize.

Maybe this is related to getting garbage data out of the ADI ports at the start of the program. Is there a way for us to poll if the ADI ports have been initialized?

HotelCalifornia, 2019-09-19:
Wondering if the solution could be as simple as swapping the order `vfs_initialize` and `vdml_initialize`. The fact that serial output is initialized before VDML could potentially explain the appearance of garbage data as @Octogonapus mentioned.

HotelCalifornia, 2019-09-19:
Actually, now I think about it, the two issues may be separate after all.

The terminal output timing issue may just be because `vfs_initialize` takes a while to start up and/or gets preempted by the competition tasks. Inserting a small delay would probably fix it, hacky as that seems. @theol0403 does this happen on kernel versions <3.2.0? or is this a new issue?

theol0403, 2019-09-20:
This appears to be a new issue. Testing on 3.1.6 gives expected behavior:

[image]

HotelCalifornia, 2019-09-20:
As far as I can verify the only thing that's changed in that neck of the woods was moving `display_initialize` to before `system_daemon_initialize`... If that's what ended up causing this then there's some pretty fragile timing at play here.

I'll try to repro this later and mess around with it a bit.

HotelCalifornia, 2019-12-18:
@theol0403 is this still happening?

WillXuCodes, 2024-12-26:
Closing this because I don't believe this is probably happening... I remember before I left we tried recreating this behavior and I don't think we were successful 

(could be completely wrong here)

closed 2024-12-26 by WillXuCodes