purduesigbots/pros #172: Project upgraded to 3.2.0 no longer runs.

SpencerJ21, 2019-09-20:
#### Expected Behavior:
It would work the same as 3.1.6 (only change was upgrading kernel and moving initialize() and opcontrol() to the new main.cpp file).

#### Actual Behavior:
After upgrading, the project would no longer run (no compilation errors nor segfaults, but nothing was being printed to the screen)(this is a screen only project)

#### Steps to reproduce:
The project in question is screenlib ([3.1.6 version](https://github.com/SpencerJ21/screenlib/tree/develop),  [3.2.0 version](https://github.com/SpencerJ21/screenlib/tree/feature/pros3.2))

Compiling, uploading, and running the 3.1.6 version works fine (the brain should print the image at the top of the README), whereas 3.2.0 does nothing.

#### System information:
V5, Kernel 3.2.0, Okapilib 3.3.13

#### Additional Information
Note, this error only occurs when Hot/Cold linking is *disabled*. I tested it out in a new, blank repository, so I'm assuming the error lies in project upgrading.

I also just checked this, the upgraded project does not have the same `common.mk` as the one in this repository, so that might be the root of this issue. I don't have access to a brain right now, but I might be able to test it later.

HotelCalifornia, 2019-09-22:
@SpencerJ21 were you able to test it after verifying `common.mk`?

SpencerJ21, 2019-09-24:
I was just able to test some things, and unfortunately just manually updating `common.mk` didn't work. However, after messing around some more, I found what did fix the issue was updating the `EXCLUDE_SRC_FROM_LIB` in `Makefile` to exclude `main.cpp` instead of the three files of the past. I thought I had done this but apparently not. Everything seems to be working fine now, thanks.