purduesigbots/pros #173: how to initialize pros::ADIEncoder ?

3038922, 2019-09-21:
```
ADIEncoder::ADIEncoder(const std::string &name, const json &pragma) : _name(name),
                                                                      _port(std::make_pair<int, int>(pragma[name].at(0).get<int>(), pragma[name].at(1).get<int>())),
                                                                      pros::ADIEncoder(pragma[name][0].get<std::uint8_t>(), pragma[name][1].get<std::uint8_t>())
{
    sysData->adiCheck(_port, _name);
    logger->info({_name, ":", std::to_string(_port.first), " ", std::to_string(_port.second), "构造成功"});
}
```
3.16 Write everything OK like this

```
  pros::ADIEncoder(1,2);
```
It's no problem to write this directly.
But my data comes from JSON.
Whether I read int or uint8_t with json, I make mistakes.
There's no problem with compilation. It crashes when it runs.

HotelCalifornia, 2019-09-21:
Are you saying there's some sort of regression in 3.2.0?

What do you mean by "it crashes"? Is there a segfault ("memory permission error")? Does it just do nothing?

3038922, 2019-09-21:
It's like falling into while (1);
I'm sure there's no problem with json's reading. It's been tested.
I made a serialized JSON. I made a graphics library. I realized the graphics of all configurations, PID parameter settings and so on. I improved the work efficiency.

HotelCalifornia, 2019-09-21:
So, to clarify again, are you saying this was introduced in the upgrade to 3.2.0?

3038922, 2019-09-21:
Yes.I now go back to 3.16, the program is running well.
Just before I used int.
`pros::ADIEncoder(pragma[name][0].get<int>(), pragma[name][1].get<int>())`
And the most amazing thing is that several V5 brains are all right. Most V5 brains have problems.
The students'development environment is the same as mine.
```
$ arm-none-eabi-g++ --version                                                                                           arm-none-eabi-g++.exe (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
```

3038922, 2019-09-21:
The reason is found. If encL is set (2,3) ;encR set(4,5) it will be dead.
Some students'enc settings are {3,4} {5,6} So they can run normally.

HotelCalifornia, 2019-09-21:
From the [documentation for `pros::ADIEncoder`](https://pros.cs.purdue.edu/v5/api/cpp/adi.html#pros-adiencoder):

Parameters |	 description
-- | --
port_top	| the “top” wire from the encoder sensor with the removable cover side UP. This should be in port 1, 3, 5, or 7 (‘A’, ‘C’, ‘E’, ‘G’).
port_bottom | the “bottom” wire from the encoder sensor

closed 2019-09-21 by HotelCalifornia