purduesigbots/pros #176: Hot/Cold build problems on Fedora 30 with GCC 9.2.0

TixoRebel, 2019-10-10:
#### Expected Behavior:
Project should build with Hot/Cold linking enabled.

#### Actual Behavior:
Project fails to build when Hot/Cold linking is enabled with the following errors:

```
Creating cold package with libpros,libc,libm [ERRORS]
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(cow-fs_ops.o): in function `std::filesystem::current_path(std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:748: undefined reference to `pathconf'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:767: undefined reference to `getcwd'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(cow-fs_ops.o): in function `std::filesystem::current_path(std::filesystem::path const&, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:804: undefined reference to `chdir'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(cow-fs_ops.o): in function `(anonymous namespace)::create_dir(std::filesystem::path const&, std::filesystem::perms, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:581: undefined reference to `mkdir'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(cow-fs_ops.o): in function `std::filesystem::permissions(std::filesystem::path const&, std::filesystem::perms, std::filesystem::perm_options, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:1132: undefined reference to `chmod'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(fs_ops.o): in function `std::filesystem::current_path(std::filesystem::__cxx11::path const&, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:804: undefined reference to `chdir'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(fs_ops.o): in function `(anonymous namespace)::create_dir(std::filesystem::__cxx11::path const&, std::filesystem::perms, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:581: undefined reference to `mkdir'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(fs_ops.o): in function `std::filesystem::permissions(std::filesystem::__cxx11::path const&, std::filesystem::perms, std::filesystem::perm_options, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:1132: undefined reference to `chmod'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(fs_ops.o): in function `std::filesystem::do_copy_file(char const*, char const*, std::filesystem::copy_options_existing_file, stat*, stat*, std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/../filesystem/ops-common.h:439: undefined reference to `chmod'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/lib/libstdc++.a(fs_ops.o): in function `std::filesystem::current_path[abi:cxx11](std::error_code&)':
/builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:748: undefined reference to `pathconf'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: /builddir/build/BUILD/arm-none-eabi-gcc-cs-9.2.0/gcc-arm-none-eabi/arm-none-eabi/libstdc++-v3/src/c++17/../../../../../gcc-9.2.0/libstdc++-v3/src/c++17/fs_ops.cc:767: undefined reference to `getcwd'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: ./firmware/libc.a(lib_a-unlinkr.o): in function `_unlink_r':
unlinkr.c:(.text._unlink_r+0x12): undefined reference to `_unlink'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: ./firmware/libc.a(lib_a-linkr.o): in function `_link_r':
linkr.c:(.text._link_r+0x16): undefined reference to `_link'
/usr/lib/gcc/arm-none-eabi/9.2.0/../../../../arm-none-eabi/bin/ld: ./firmware/libc.a(lib_a-statr.o): in function `_stat_r':
statr.c:(.text._stat_r+0x16): undefined reference to `_stat'
collect2: error: ld returned 1 exit status
```

#### Steps to reproduce:
Create a new project with our without Okapi and optionally remove the starting LLEMU code from `main.cpp`. Disabling Hot/Cold linking will allow the project to build as expected.

#### System information:
Platform: V5
PROS Kernel Version: 3.2.0
Operating System: Fedora 30
GCC Version: 9.2.0

nickmertin, 2019-10-12:
Verified on the same setup on Arch Linux.

luigishat, 2019-10-23:
I'm getting this with Arch Linux with  pros kernel 3.2.0.

sazrocks, 2019-11-02:
Also having the same issue on Arch with toolchain 9.2.0 and kernel 3.2.0.

srithon, 2019-11-09:
Same issue on Arch
arm-none-eabi-gcc v9.2.0
arm-none-eabi-binutils v2.33.1-1 (dependency)
arm-none-eabi-newlib v3.1.0-2
Kernel v3.2.0

HotelCalifornia, 2019-11-10:
Temporary workaround for this is to just define stubs for each of the symbols it says it can't find:

- `chdir`
- `mkdir`
- `chmod`
- `pathconf`
- `getcwd`
- `_unlink`
- `_link`
- `_stat`

(you'll have to verify the signatures for each of these functions of course)

deweykai, 2019-12-06:
Also having this issue on mac Catalina.
arm-none-eabi-gcc v9.2.1

HotelCalifornia, 2019-12-06:
#184

brizuelasamuelito, 2020-01-18:
> Temporary workaround for this is to just define stubs for each of the symbols it says it can't find:
> 
> * `chdir`
> * `mkdir`
> * `chmod`
> * `pathconf`
> * `getcwd`
> * `_unlink`
> * `_link`
> * `_stat`
> 
> (you'll have to verify the signatures for each of these functions of course)

How do you define stubs?

nathan-moore, 2020-01-18:
https://en.wikipedia.org/wiki/Method_stub

Essentially you define some method to just make the symbol to satisfy the linker. The easier way would be to just disable hot/cold.

closed 2020-01-23 by TixoRebel

lerker100, 2020-06-05:
I'm getting this error on Manjaro with GCC 9.3 and wanted to make sure that this is fixed with this issue.
```
Creating cold package with libc libm libpros okapilib, [ERRORS]
arm-none-eabi-g++: error: unrecognized command line option '--whole-archive'; did you mean '--whole-file'?
make: *** [common.mk:230: bin/cold.package.elf] Error 1
arm-none-eabi-g++: error: unrecognized command line option '--no-whole-archive,'; did you mean '--no-whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--gc-sections'; did you mean '--data-sections'?
arm-none-eabi-g++: error: unrecognized command line option '--start-group'
arm-none-eabi-g++: error: unrecognized command line option '--end-group'
Error: Failed to build
```

HotelCalifornia, 2020-06-05:
@lerker100 that's #215