purduesigbots/pros #183: Prefetch error when running code via timed run after compiling with GCC 9

gnations56, 2019-11-11:
#### Expected Behavior:

Autonomous to run
#### Actual Behavior:

"Prefetch error! 0000000" when running timed run
"DATA ABORT EXCEPTION PC: 3828996" when using the comp switch
#### Steps to reproduce:

#### System information:
Operating System:  
Ubuntu 18.04
PROS Version: 
3.1.4
#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
[image]

gnations56, 2019-11-11:
When I compile the same exact code using GCC 7.2 it works

gnations56, 2019-11-11:
I can try later with a minimal repo and gcc 9

nathan-moore, 2019-11-25:
@gnations56 did you ever figure this out?

HotelCalifornia, 2019-11-26:
I'm looking at this again, and the picture of the screen doesn't seem to match up with the report-- it shows a data abort not a prefetch abort.

nathan-moore, 2019-12-04:
I'm going ahead and closing this. If you ever come up with a repo or some way we can diagnose, feel free to re-open.

closed 2019-12-04 by nathan-moore