purduesigbots/pros #186: Hot/Cold uploading does not work

daniel-sudz, 2019-11-25:
#### Expected Behavior:
For Hot/Cold reloading to work 

#### Actual Behavior:
Hot/Cold reloading does not work 

#### Steps to reproduce:
To enable hot cold reploading and compile the project with `prosv5 make` using the latest binary and pros toolchain on windows 

Everything builds fine, but when I run program, I see empty screen (it’s sort of grayish, but closer to black).
Deleting all programs from Cortex does not help.
I believe the issue is – hot.package.bin is essentially empty (413 bytes).

#### System information:
Windows 10
Latest firmware and toolchain 

PROS Version: version 3.1.4

#### Additional Information
Here are the sizes of key files after building:
           872,092 cold.package.bin
         2,014,944 cold.package.elf
           273,524 program.cpp.o
               413 hot.package.bin
           462,036 hot.package.elf

#### Screenshots/Output Dumps/Stack Traces

daniel-sudz, 2019-11-25:
I am facing this issue with a bare project and freshly installed prosv5. I might be able to provide a full reproducible VM image in the next couple of days. Based on the hot package size though it is clear that no user code is being included.

nathan-moore, 2019-11-25:
What version of the kernel are you using?

Octogonapus, 2019-11-25:
@daniel-sudz Does it work if you switch back to monolith?

nathan-moore, 2019-11-25:
I also doubt that a VM image is necessary. You could likely just give us a zip of the project.

daniel-sudz, 2019-11-26:
The problem was resolved. My issue was missing export C commands. Resolved by doing this: 
```
extern "C" {
   void autonomous();
   void initialize();
   void disabled(void);
   void competition_initialize(void);
   void opcontrol();
}
```

closed 2019-11-26 by daniel-sudz