purduesigbots/pros #194: LVGL lv_kb not included in API

QSSaber, 2020-01-09:
#### Expected Behavior:
It should compile without problem as it is a function supported since LVGL 5.1. 

#### Actual Behavior:
lv_kb command set is not included within the LVGL API in pros.
Cannot call lv_kb functions 
Error in compiling 

#### Steps to reproduce:

any lv_kb functions 

By changing include/display/lv_conf.h line 296 #define USE_LV_KB 0 to 1, it can be compiled but fails to build output.bin, blocked execution by collect2.exe

#### System information:
V5 
Kernel Version 3.2.0
Windows 10 LTSC

#### Additional Information
Terminal (w/o change in lv_conf.h)
D:\19WS_3IT>prosv5 build
Compiled src/initialize.cpp [WARNINGS]
src/initialize.cpp: In function 'lv_res_t btnAction(lv_obj_t*)':
src/initialize.cpp:120:40: error: 'lv_kb_create' was not declared in this scope
                         lv_obj_t *kb = lv_kb_create(lv_scr_act(), NULL);
                                        ^~~~~~~~~~~~
src/initialize.cpp:120:40: note: suggested alternative: 'lv_cb_create'
                         lv_obj_t *kb = lv_kb_create(lv_scr_act(), NULL);
                                        ^~~~~~~~~~~~
                                        lv_cb_create
src/initialize.cpp:121:25: error: 'lv_kb_set_cursor_manage' was not declared in this scope
                         lv_kb_set_cursor_manage(kb, true);
                         ^~~~~~~~~~~~~~~~~~~~~~~
src/initialize.cpp:121:25: note: suggested alternative: 'lv_ta_set_cursor_type'
                         lv_kb_set_cursor_manage(kb, true);
                         ^~~~~~~~~~~~~~~~~~~~~~~
                         lv_ta_set_cursor_type
src/initialize.cpp:122:45: error: 'LV_KB_STYLE_BG' was not declared in this scope
                         lv_kb_set_style(kb, LV_KB_STYLE_BG, &lv_style_transp_tight);
                                             ^~~~~~~~~~~~~~
src/initialize.cpp:122:45: note: suggested alternative: 'LV_CB_STYLE_BG'
                         lv_kb_set_style(kb, LV_KB_STYLE_BG, &lv_style_transp_tight);
                                             ^~~~~~~~~~~~~~
                                             LV_CB_STYLE_BG
src/initialize.cpp:122:25: error: 'lv_kb_set_style' was not declared in this scope
                         lv_kb_set_style(kb, LV_KB_STYLE_BG, &lv_style_transp_tight);
                         ^~~~~~~~~~~~~~~
src/initialize.cpp:122:25: note: suggested alternative: 'lv_cb_set_style'
                         lv_kb_set_style(kb, LV_KB_STYLE_BG, &lv_style_transp_tight);
                         ^~~~~~~~~~~~~~~
                         lv_cb_set_style
src/initialize.cpp:123:45: error: 'LV_KB_STYLE_BTN_REL' was not declared in this scope
                         lv_kb_set_style(kb, LV_KB_STYLE_BTN_REL, &rel_kb);
                                             ^~~~~~~~~~~~~~~~~~~
src/initialize.cpp:123:45: note: suggested alternative: 'LV_MBOX_STYLE_BTN_REL'
                         lv_kb_set_style(kb, LV_KB_STYLE_BTN_REL, &rel_kb);
                                             ^~~~~~~~~~~~~~~~~~~
                                             LV_MBOX_STYLE_BTN_REL
src/initialize.cpp:124:45: error: 'LV_KB_STYLE_BTN_PR' was not declared in this scope
                         lv_kb_set_style(kb, LV_KB_STYLE_BTN_PR, &pr_kb);
                                             ^~~~~~~~~~~~~~~~~~
src/initialize.cpp:124:45: note: suggested alternative: 'LV_MBOX_STYLE_BTN_PR'
                         lv_kb_set_style(kb, LV_KB_STYLE_BTN_PR, &pr_kb);
                                             ^~~~~~~~~~~~~~~~~~
                                             LV_MBOX_STYLE_BTN_PR
src/initialize.cpp:125:44: error: 'LV_KB_MODE_NUM' was not declared in this scope
                         lv_kb_set_mode(kb, LV_KB_MODE_NUM);
                                            ^~~~~~~~~~~~~~
src/initialize.cpp:125:44: note: suggested alternative: 'LV_SB_MODE_AUTO'
                         lv_kb_set_mode(kb, LV_KB_MODE_NUM);
                                            ^~~~~~~~~~~~~~
                                            LV_SB_MODE_AUTO
src/initialize.cpp:125:25: error: 'lv_kb_set_mode' was not declared in this scope
                         lv_kb_set_mode(kb, LV_KB_MODE_NUM);
                         ^~~~~~~~~~~~~~
src/initialize.cpp:125:25: note: suggested alternative: 'lv_ta_set_sb_mode'
                         lv_kb_set_mode(kb, LV_KB_MODE_NUM);
                         ^~~~~~~~~~~~~~
                         lv_ta_set_sb_mode
src/initialize.cpp:126:25: error: 'lv_kb_set_ok_action' was not declared in this scope
                         lv_kb_set_ok_action(kb, inputHandler);
                         ^~~~~~~~~~~~~~~~~~~
src/initialize.cpp:126:25: note: suggested alternative: 'lv_cb_set_action'
                         lv_kb_set_ok_action(kb, inputHandler);
                         ^~~~~~~~~~~~~~~~~~~
                         lv_cb_set_action
src/initialize.cpp:138:25: error: 'lv_kb_set_ta' was not declared in this scope
                         lv_kb_set_ta(kb, ta);
                         ^~~~~~~~~~~~
src/initialize.cpp:138:25: note: suggested alternative: 'lv_cb_set_text'
                         lv_kb_set_ta(kb, ta);
                         ^~~~~~~~~~~~
                         lv_cb_set_text
mv: cannot stat '.d/initialize.Td': No such file or directory
make: *** [common.mk:269: bin/initialize.cpp.o] Error 1
Capturing metadata for PROS Editor...
Error: Failed to build

w/ lv_conf.h change

Compiled src/autonomous.cpp [OK]
Compiled src/initialize.cpp [OK]
Compiled src/opcontrol.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and okapilib,libm,libc,libpros [ERRORS]
bin/initialize.cpp.o: In function `btnAction(_lv_obj_t*)':
D:\19WS_3IT/src/initialize.cpp:120: undefined reference to `lv_kb_create'
D:\19WS_3IT/src/initialize.cpp:121: undefined reference to `lv_kb_set_cursor_manage'
D:\19WS_3IT/src/initialize.cpp:122: undefined reference to `lv_kb_set_style'
D:\19WS_3IT/src/initialize.cpp:123: undefined reference to `lv_kb_set_style'
D:\19WS_3IT/src/initialize.cpp:124: undefined reference to `lv_kb_set_style'
D:\19WS_3IT/src/initialize.cpp:125: undefined reference to `lv_kb_set_mode'
D:\19WS_3IT/src/initialize.cpp:126: undefined reference to `lv_kb_set_ok_action'
D:\19WS_3IT/src/initialize.cpp:138: undefined reference to `lv_kb_set_ta'
collect2.exe: error: ld returned 1 exit status
make: *** [common.mk:237: bin/hot.package.elf] Error 1
Capturing metadata for PROS Editor...
Error: Failed to build

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2020-01-10:
Yeah you can't really change lv_conf from your own code, as LVGL is precompiled into libpros.a with whatever settings I gave it. I guess I never saw a reason for enabling the keyboard, but I can add it for the next release.

QSSaber, 2020-01-10:
Thanks @HotelCalifornia

PR #198 (2020-01-21): ✨ Enable LVGL keyboard
#### Summary:

does what it says on the tin

#### Motivation:

someone asked for it

##### References (optional):

closes #194 

#### Test Plan:

- [x] program with lv_kb compiles and works
closed 2020-01-21 by nathan-moore