purduesigbots/pros #205: calling imu get_rotation() in pros::lcd::print results in jump table errors

22pilarskil, 2020-01-29:
#### Expected Behavior:

The sensor value should be displayed to the lcd screen

#### Actual Behavior:

Nothing is being registered- over the lcd, there is an error message reading "Jump Table error ! 03813374"

#### Steps to reproduce:

Update prosv5 conductor to latest version (3.2.1) and initialize an Imu sensor. Call get_rotation() during opcontrol.

#### System information:
Platform: 
PROS Kernel Version: 

#### Additional Information

kernel version: 3.2.1 (pros-mainline)
okapilib version:4.0.2 (pros-mainline)
#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2020-01-29:
Make sure you're using vexos version 1.0.9

nickmertin, 2020-01-29:
I'm not entirely aware of how the relevant internals work, but if possible it would be nice to be able to return a value such as `PROS_ERR_F` in situations like this.

nathan-moore, 2020-01-29:
And throw a not supported(Upgrade vexos) in c++ (or equivalent)? I agree, expecially with how 1.0.8 appears to work, but is buggy.

nickmertin, 2020-01-29:
Are exceptions the current method of reporting issues in the C++ API? From my understanding, the C++ APIs also return `PROS_ERR` or `PROS_ERR_F` and set `errno` appropriately, rather than using exceptions support.

HotelCalifornia, 2020-01-29:
Warning users to upgrade their firmware is a to-do project for the CLI. As far as gating features to firmware versions:
- I was told that backwards compatibility has become less of a concern for vexos recently (thus jump table faults when using firmware <1.0.8)
- we could certainly do version checking similar to what was implemented in the controller LCD clear command

> Are exceptions the current method of reporting issues in the C++ API?
no, but presumably we could fail early in constructors

Octogonapus, 2020-01-29:
> we could fail early in constructors

This is OkapiLib's preferred method for fatal errors.

22pilarskil, 2020-01-31:
Thanks a lot guys! It was my VEXos version- its 1.0.9, everything works smoothly. It was very confusing but for anyone who needs to upgrade their prosv5 CLI and VEXos firmware,

1) Visit this link and download the firmware updater: https://www.vexrobotics.com/vexedr/products/firmware
2) Go to terminal and type prosv5 conductor update (or upgrade, i forget)

Thanks again!

HotelCalifornia, 2020-01-31:
`prosv5 conductor upgrade` does not upgrade the CLI, rather it deals with all of a project's installed templates. The only reason that step should have been necessary is that you weren't already on the latest version of the kernel which is where we added support for the VEX Inertial sensor (which I don't think was the case because you wouldn't have been able to use the IMU class at all if it was).

In any case, I'm glad this problem was resolved.

closed 2020-01-31 by HotelCalifornia