purduesigbots/pros #206: Recieving of errors when compiling project

mkoujiri, 2020-02-01:
#### Expected Behavior:
Project Compiled

#### Actual Behavior:
recieved a list of errors after the prompt: 
Creating cold package with libc libm libpros okapilib, [ERRORS]
arm-none-eabi-g++: error: unrecognized command line option '--whole-archive'; did you mean '--whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--no-whole-archive,'; did you mean '--no-whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--gc-sections'; did you mean '--data-sections'?
arm-none-eabi-g++: error: unrecognized command line option '--start-group'
arm-none-eabi-g++: error: unrecognized command line option '--end-group'

#### Steps to reproduce:
Compile with make.

#### System information:
Void Linux

3.1.4

#### Additional Information
I've made this work on void linux in the past, I never recieved this issue

#### Screenshots/Output Dumps/Stack Traces

nathan-moore, 2020-02-01:
I assume this is an unmodified common.mk? Can you post the g++ command that's failing for you? You will have to add VERBOSE=1 when you invoke make to see the full command. 
This will happen if linker flags are passed to arm-none-eabi-g++ instead of the linker. I double checked and we're prefixing these flags with -Wl, so I'm not sure why this is happening.

mkoujiri, 2020-02-02:
When running with verbose, I recieve this: 
```arm-none-eabi-g++ -mcpu=cortex-a9 -mfpu=neon-fp16 -mfloat-abi=softfp -Os -g -Wno-psabi -nostdlib -ffunction-sections -fdata-sections -fdiagnostics-color -funwind-tables -Wl,--gc-keep-exported --whole-archive firmware/libpros.a firmware/okapilib.a -lstdc++ --no-whole-archive, -Wl,-T./firmware/v5.ld --gc-sections --start-group ./firmware/libc.a ./firmware/libm.a ./firmware/libpros.a ./firmware/okapilib.a -lc -lm -lgcc -lstdc++ -lsupc++ --end-group -o bin/cold.package.elf,
Creating cold package with libc libm libpros okapilib, [ERRORS]
arm-none-eabi-g++: error: unrecognized command line option '--whole-archive'; did you mean '--whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--no-whole-archive,'; did you mean '--no-whole-file'?
arm-none-eabi-g++: error: unrecognized command line option '--gc-sections'; did you mean '--data-sections'?
arm-none-eabi-g++: error: unrecognized command line option '--start-group'
arm-none-eabi-g++: error: unrecognized command line option '--end-group'
make: *** [common.mk:230: bin/cold.package.elf] Error 1
```

I haven't modified the common.mk at all

HotelCalifornia, 2020-02-02:
1. What version of arm-none-eabi-gcc do you have installed? Run it with `--version`
2. How did you install it?

nathan-moore, 2020-02-02:
> -Wl,-T./firmware/v5.ld --gc-sections --start-group ./firmware/libc.a ./firmware/libm.a ./firmware/libpros.a ./firmware/okapilib.a -lc -lm -lgcc -lstdc++ -lsupc++ --end-group

That's what's causing this problem. I updated to gnu-make 4.3 and am able to repo this. In the meantime, I know 4.1 works fine.

HotelCalifornia, 2020-02-02:
I'm confused, how does gnu-make have any bearing on what flags the compiler can accept?

nathan-moore, 2020-02-02:
The flags are the same, but they should be comma separated. It should look something like 

-Wl,-T./firmware/v5.ld,--gc-sections,--start-group ect...

My guess is that how we set the SPACE variable is broken, which breaks where we comma separate them (https://github.com/purduesigbots/pros/blob/1e07ae915b38689ae53a9f03cacc8fbbed09dbff/common.mk#L24)

mkoujiri, 2020-02-02:
```
Previously appending using '+=' to an empty variable would result in a value
  starting with a space.  Now the initial space is only added if the variable
  already contains some value.  Similarly, appending an empty string does not
  add a trailing space.```
I'm not sure how creating a space variable is possible anymore.  I will attempt to research it.

PR #207 (2020-02-02): 🥅 Fix Makefile spaces definition
#### Summary:
Gnu Make 4.3 broke how we generated spaces. This fixes that.

#### Motivation:
Fix pros for people who actually keep things updated

##### References:
fixes #206 

#### Test Plan:

- [x] compiles under 4.3
- [x] compiles under earlier versions
closed 2020-02-02 by HotelCalifornia